FutureSDR/FutureSDR #3: ZeroMQ PubSink & SubSource

loic-fejoz, 2021-08-29:
**Why ZeroMQ?**
ZeroMQ is by default in GNU Radio Companion.

**ZMQSubSource & ZMQPubSink as blocks**
I would like to have blocks to publish stream & receive stream from other systems.

**Underlying ZMQ rust lib?**
I have done [an experiment](https://github.com/loic-fejoz/FutureSDR/tree/feature/ZeroMQ) with [zmq](https://crates.io/crates/zmq) library which is blocking one.
Should we consider using [async-zmq](https://crates.io/crates/async-zmq) (non blocking) or the more idiomatic [libzmq](https://crates.io/crates/libzmq) (but unmaintained).

**Additional context**
Having those would enable tests where only part of a chain is migrated to FutureSDR. For instance, it could enable to split a chain between GNU Radio and FutureSDR until FutureSDR  had all the mandatory blocks of the overall chain.

bastibl, 2021-08-30:
I'll have a close look at the implementation, as soon as I have time. But just because I forgot to mention it in the intro video: One can specify through the BlockMeta that a block uses blocking calls in work, which will spawn it in a separate thread, using its own executor. See
https://docs.rs/futuresdr/0.0.4/futuresdr/runtime/struct.BlockMetaBuilder.html#method.blocking
https://github.com/FutureSDR/FutureSDR/blob/master/src/blocks/soapy_src.rs#L27

loic-fejoz, 2021-08-30:
@bastibl Thanks for the tip. I will add this.
Once I have properly add ZeroMQ as a feature (by default?), I will create a pull request.

PR #9 (2021-09-02): feature/ZeroMQ
Add ZMQSubSource and SMQPubSink for streaming over ZeroMQ protocol which is also by default in GNU Radio.
I think I have made it as a feature activated by default.
It also come with two binaries (one receiver, one emitter) that can talk to each other and/or with GNU Radio.

Apparently the topics is not settable in GNU Radio thus I have not provided way to set it. Maybe I should?

Close #3
closed 2021-09-02 by bastibl