FutureSDR/FutureSDR #22: Produce More Output

wspeirs, 2021-12-08:
I'm working on a FIR filter, and need the ability to produce more output than the length of the output slice. The pattern _seems_ to be:

```
        let i = sio.input(0).slice::<f32>();
        let o = sio.output(0).slice::<f32>();

        let n = cmp::min(i.len(), o.len());

... do work ...

        sio.output(0).produce(n);
```

Then only work on `n` pieces of input/output. However, given interpolation for the FIR filter, it might produce more output than input. If I had say `n*2` output, how can I make that work?

From my understanding of the code, there is _no_ interface to do that.

bastibl, 2021-12-09:
Yes, but that's as intended: you'd have to do: `let n = cmp::min(i.len() + 1 - taps.len(), o.len());`

Here is a FIR filter with some performance tweaks that require nightly:
https://github.com/bastibl/FutureSDR/blob/master/src/blocks/fir.rs

Performs very well on my system.

wspeirs, 2021-12-09:
@bastibl thanks for the reply! What about an interpolating FIR filter? Or an interpolating resampler? That would produce "more" output than input... how should I handle that? Thanks!

bastibl, 2021-12-09:
Ah, now I understand the problem: You always get access to all buffer space that is currently available. But consuming input items is independent from producing output items. So when you have an interpolation of 2 and 20 input items and 10 output items, you would consume only 5 of the 20 input items and produce 10 items.

wspeirs, 2021-12-09:
Ah, understood... thanks!

Along these lines, I was looking at the [discussion](https://github.com/FutureSDR/FutureSDR/discussions/6) about blocks to add, and had previously created an interpolating & decimating FIR & IIR in C++, and was looking to convert them to this project. Given you already have a FIR in your fork, what is the best path forward to prevent duplicitous work? Is there an updated Wiki/Discussion for the blocks to work on? I have some free hours today, and would love to contribute... thanks!

bastibl, 2021-12-09:
I implemented the filter in the fork for performance measurements to compare FutureSDR to GNU Radio.

I have no strong opinion regarding the integration of your filters. (Would be really cool btw!) If you want and if that works for you, you could use the other filter as a base (adding interpolation as an option, for example). But there could just as well be separate interpolating/decimating filters.

wspeirs, 2021-12-09:
I see you contribute to GNU Radio... what would be the best way to _easily_ compare results? Basically I want to specify an interpolation, taps, and inputs to `interp_fir_filter_fff`, then get the output. I'm struggling to do this in both C++ and Python. Thoughts on how to easily generate a few tests cases to ensure my code is working properly? Thanks!

bastibl, 2021-12-10:
You can create similar flowgraphs for FutureSDR and GNU Radio that read data from a file source, process it with the same filter, write results to a file. With GNU Radio, you can easily create these flowgraph with GRC. You can add parameters with the Parameter block. This way you can give the flowgraph command line parameters to set the input/output file, for example.
For FutureSDR, you can see examples in the `perf` directory to see how command line parameters can be used.
Note that both read and write binary files. You can then read and parse the file with any scripting language and look at the differences.

bastibl, 2021-12-11:
I'll close the issue for know. If there are any more question, it can be converted to a discussion.

closed 2021-12-11 by bastibl