FutureSDR/FutureSDR #27: FileSink repeated samples when writing to disk

lkolbly, 2022-01-29:
So, I setup a graph which generates a frequency sweep (about the limit of my signal processing knowledge), throttles it, and then writes it to file:
```rust
fn main() -> Result<()> {
    let mut fg = Flowgraph::new();

    // Make a chirp source
    let mut freq = 0.0;
    let mut current_value = Complex::from_polar(1.0, 0.0);
    let src = fg.add_block(Source::<f32>::new(move || {
        freq += 0.00003;
        if freq > 3.14159265 / 2.0 {
            freq = 0.0;
        }
        current_value = current_value * Complex::from_polar(1.0, freq);
        current_value.re
    }));

    // Throttle that & send it to file
    let throttle = fg.add_block(Throttle::new(4, 100_000.0));
    let sink = fg.add_block(FileSink::new(4, "/tmp/tmp.f32"));

    fg.connect_stream(src, "out", throttle, "in")?;
    fg.connect_stream(throttle, "out", sink, "in")?;

    Runtime::new().run(fg)?;

    Ok(())
}
```
(I've pushed a branch which reproduces the problem here: https://github.com/lkolbly/FutureSDR/tree/lkolbly/clicks)

when I view the resulting file `/tmp/tmp.f32` in inspectrum, I see this:
[image]
where I would expect a smooth tone sweep up to the top, starting again at the "bottom" (middle). To my untrained eye this looks like the `FileSink` block is repeatedly re-writing the last "few" samples, while it waits for the source + throttle to generate the next block of samples. (notably, I also have a somewhat more complicated flow graph involving FIR filters and such, which shows a similar behaviour but the glitches are further apart, maybe because the system is under more load?)

I considered the possibility that this was because of `FileSink` ignoring the item size, but I also wrote up [this commit](https://github.com/lkolbly/FutureSDR/commit/30941302804adbf1a0a628cbb09dcc053b833e2e) that should fix that, and I still saw this behaviour.

Because of the throttle block, it doesn't strike me as an issue with disk speed (additionally, my `/tmp` drive is on a SSD, so it should be very speedy)

It's also entirely possible that I'm just doing something dumb - but, is this known/unexpected behaviour?

loic-fejoz, 2022-01-30:
I have checkout your branch.
First comment: it does not compile but whatever. I have modify it to just generate u8 so that I do not need to change FileSink and every items in the graph is of size 1.
```
(current_value.re * 255.0) as u8;
```

Then I try with and without throtlle and the result is exact same (but obviously without throtlle it fills the file faster).

Then I check with a python script to really understand what as going on:
```
with open("/tmp/tmp.u8", "rb") as binary_file:
    data = binary_file.read(600000)
    data = [float(c) for c in data]
    print(data)
    import numpy as np
    import matplotlib.pyplot as plt

    x = np.array(np.arange(0, len(data)))
    y = np.array(data)
    plt.plot(x, y)

    plt.show() # affiche la figure a l'ecran
```
And finally, as I am not sure how to understand 

>  a smooth tone sweep up

I have update the code to

```
current_value = Complex::from_polar(1.0, angle);
let res: u8 = (current_value.im * 255.0) as u8;
```

And I got this plot in the temporal domain:
[image]

Obviously Inspectrum then display a complete different frequency domain view.

I hope it helps to understand what is going on...

lkolbly, 2022-01-30:
Aha, yes! Thank you, after looking through the time domain data a bit more it looks like it's because `FileSink` actually doesn't support non-size-1 item sizes due to the interaction with the streams. So this is not a bug after all.

closed 2022-01-30 by lkolbly

loic-fejoz, 2022-01-30:
But for sure, a typed FileSink would be much better.
One should create a ticket for this ...and the implementation! :speak_no_evil:

lkolbly, 2022-01-30:
That's not a bad idea: https://github.com/FutureSDR/FutureSDR/pull/28