FutureSDR/FutureSDR #44: Add signal source block

AndersKaloer, 2022-03-14:
It is often useful to have a source block that can generate basic functions, such as sinusoids, sawtooths, etc. While this can be done with the `Source<A>`, it is not particularly elegant and requires unsafe code due to potential race conditions:
```rust
static mut T: usize = 0;
let src = Source::<f32>::new(|| {
    let x = unsafe {
        T += 1;
        T
    };
    (2.0 * std::f32::consts::PI * (x as f32) * 0.1).sin()
});
```

A simple way to circumvent this issue would be to add a counter as argument the closure, so that it changes from `f: impl FnMut() -> A + Send + 'static` to `f: impl Fn(usize) -> T + Send + 'static`.

The implementation is straightforward, and I would be happy to create the block as well as some standard functions. The main question is whether `Source<A>` should be modified to include a counter, or should be implemented in a new `SignalSource<A>` block?

lkolbly, 2022-03-14:
Would it be possible to store the state of the counter inside the closure? Something like:
```rust
let mut cnt: usize = 0;
let src = Source::<f32>::new(move || {
    cnt = cnt + 1;
    (2.0 * std::f32::consts::PI * (cnt as f32) * 0.1).sin()
});
```
(not at my computer right now so can't test the exact syntax)

AndersKaloer, 2022-03-14:
Yes, that works, thanks! Then the current block should be fine, so I will close this issue.

closed 2022-03-14 by AndersKaloer