FutureSDR/FutureSDR #47: Document required dependencies

ratzrattillo, 2022-03-22:
On different operating systems different dependencies are required to make FutureSDR run smoothly with common SDR hardware.

Compiling the FutureSDR examples on Windows e.g. fails when compiling sopaysdr-sys 0.7.3:
`pkg_config: Could not run "pkg-config" "--libs" "--cflags" "SoapySDR" "SoapySDR >= 0.6.0"
Most likely, you need to install a pkg-config package for your OS.`

or:

`[INFO] [UHD] Win32; Microsoft Visual C++ version 14.2; Boost_107500; UHD_4.0.0.HEAD-0-g90ce6062
Found Rafael Micro R820T/2 tuner
FutureSDR: ERROR - SoapySDR::Device::enumerate(sdrplay) sdrplay_api_Open() failed
FutureSDR: INFO - Using format CF32.`

The Readme.md should contain some information on the required dependencies e.g.:

### Dependencies for Compiling on Microsoft Windows
The following dependencies are required to compile the FutureSDR examples on Windows:

- PhotosSDR: https://downloads.myriadrf.org/builds/PothosSDR/
- LLVM Stable: https://github.com/llvm/llvm-project/releases
- SDRuno: https://www.sdrplay.com/sdruno/

Always select the "Add to PATH variable option in the installers", so the required libraries and binaries are found when compiling FutureSDR programs.
To avoid performance issues and warnings like `FutureSDR: WARN - SoapyVOLKConverters: no VOLK config file found. Run volk_profile for best performance.` please run: `volk_profile` on your commandline.

### Dependencies for Compiling on Linux
`...`
### Dependencies for Compiling on macOS
`...`

### Note:

These dependencies might change over time and should constantly be updated. It would be best to keep the dependencies to a minimum and choose the most slim packages which fulfill the minimum requirements.

loic-fejoz, 2022-03-22:
Hi,
Could you explain how you came up with those dependencies please? 
Last time, I have built on MS/Windows, I did not have such dependencies...

Moreover, if you look at [the action running cargo test under MS/Windows](https://github.com/FutureSDR/FutureSDR/runs/5610364252?check_suite_focus=true) it does not require them too.

So, in any case, those are dependencies for a particular example, not FutureSDR crate. Thus I would propose to update individual README of each examples. Especially some example are really dedicated to specific hardware and/or test.

loic-fejoz, 2022-03-22:
Or maybe you want to document how to make it work with a particular setup?

Concerning LLVM, if I remember well, this is due to some dependencies on bindgen.
So maybe just redirect to https://rust-lang.github.io/rust-bindgen/requirements.html ?

ratzrattillo, 2022-03-22:
@loic-fejoz 
It could be a good idea to directly redirect to https://rust-lang.github.io/rust-bindgen/requirements.html for the LLVM dependency.
Without SDRuno i got: ERROR - SoapySDR::Device::enumerate(sdrplay) sdrplay_api_Open() failed FutureSDR

I currently have an RTL-SDR dongle in use. The dependencies originated from trying to get the examples to work by asking for help in the discord channel.

It might make sense to document the dependencies in the individual examples.

bastibl, 2022-03-29:
I just uploaded the website code: https://github.com/FutureSDR/Website. Should be possible to build it with `npm install`, `zola serve`. Some initial info from this issue was added here: https://github.com/FutureSDR/Website/blob/main/content/learn/_index.md, which shows up here: https://www.futuresdr.org/learn/.

What do you think about this? I plan to add more information and tutorials about specific-stuff, like running FutureSDR on Android or Xilinx Zynq platforms.

ratzrattillo, 2022-03-30:
Hello Basti,

that is exactly what i was looking for. I will test out the install instructions later and see if everything works fine.
Adding some device specific instructions is a very good idea aswell. It helps newcomers to get started and not get frustrated from the beginning.

ratzrattillo, 2022-04-01:
Added pull request in the Website repository: https://github.com/FutureSDR/Website/pull/1
This fixes this issue partially. But the documentation has to grow over time.

JuliDi, 2022-04-04:
I might be able to contribute some documentation on the macOS side. However, I am a little confused by how I am supposed to run FutureSDR at the moment.
E.g. if I get the makefile in the `frontend` folder to run, nothing happens (I have no experience with gulp and nodejs, not sure whether anything should happen here at all – I would expect a browser window to open up or so).

So if you could give me some more information or steps to get some minimum working example running, I'd be happy to add some doc for macOS (at least for limeSDR and RTL).

JuliDi, 2022-04-06:
Gave it a try on Ubuntu 20.04, and @ratzrattillo's documentation seems to be fine.
However, there is no documentation for building/running the `frontend` – `npm install` gives me a huge list of errors, starting with something in node-sass. @bastibl could you please provide some hints on how to build and run it?

bastibl, 2022-04-11:
Thanks for giving it a try! Maybe the npm issue was related to the deprecated `node-sass` module. I just pushed a commit that updates it and that works for me with the latest node.js version. Overall the workflow would be
- install npm and node.js
- run `npm install` once
- install `wasm-pack` (for example with `cargo install wasm-pack`)
- once in the beginning and every time you update html, assets, or css, you run `./gulp`
- once in the beginning and every time you update Rust frontend stuff, you run `make`

Both `./gulp` and `make` put their output in the `dist` folder. If you run a flowgraph with `ctrlport_enable = true` in your config (also enabled by default for debug builds), it should search this location by default and serve it.

JuliDi, 2022-04-13:
Thanks for the instructions, using the latest commit it seems to work with `npm install` (on Ubuntu at least) and I got the wasm example running.
However the `spectrum` example seems to have some issues with the dependencies. Not sure whether this is a bug on its own or some missing dependencies on my side. If it's a bug, I can open a new issue for that, of course.
```
   Compiling spectrum v0.1.0 (<...>/FutureSDR/examples/spectrum)
error[E0425]: cannot find function `new` in module `wgpu`
  --> examples/spectrum/src/wasm.rs:28:34
   |
28 |     let log = fg.add_block(wgpu::new(broker, 2048));
   |                                  ^^^ not found in `wgpu`
```

Once the examples are running, it might be good to have them documented either in a readme or on the website. They seem like a nice way to get started.

bastibl, 2022-04-14:
I just pushed a fix to change it. Note that the wasm version of this module is a quite complex example. GPU accelerated DSP in the browser requires Google Chrome unstable/canary build or similar.

JuliDi, 2022-04-14:
Great, thanks a lot! Now it works like a charm, also on macOS – I even got a blue-ish waterfall on the spectrum example using Brave Browser (on stable I don't get any errors... seems ok). Currently I don't have the time to play around with my limesdr so I cannot say anything about the drivers yet.

Regarding the documentation: it does not look like there is anything special for macos, installing Rust etc. is pretty straightforward and a link to the website should suffice.
I recall that there was one error that I could fix by installing an older node version but that might have been fixed with your latest commits anyway.

bastibl, 2022-04-14:
Thanks for the feedback. I'd say we close the issue for now. Feel free to open other issues about specific problems.

closed 2022-04-14 by bastibl