FutureSDR/FutureSDR #49: AudioSink add complex input or multiple taps

ratzrattillo, 2022-04-06:
When using the AudioSink block with two channels, i would expect the block to have two input taps (e.g. one for the left and one for the right channel).
@loic-fejoz suggested two have either one tap per channel, or one tap accepting a Complex number as input.

Currently the AudioSink in blocks/audio/audio_sink.rs:40 creates the stream with a hardcoded size of 4 Bytes per sample:

`StreamIoBuilder::new().add_input("in", 4).build(),`

To play an audio file with only one channel (like the rick.mp3 example) on a windows PC, which in many cases only show an audio output device with two channels, hacks like this have to be done:

`
use cpal::traits::{DeviceTrait, HostTrait};
use futuresdr::anyhow::Result;
use futuresdr::blocks::ApplyIntoIter;
use futuresdr::blocks::audio::AudioSink;
use futuresdr::blocks::audio::FileSource;
use futuresdr::runtime::{Flowgraph, Runtime};

fn main() -> Result<()> {

    let host = cpal::default_host();
    let audio_device = host.default_output_device().expect("no output device available");
    println!("Device name: {}", audio_device.name().unwrap());
    let supported_configs_range = audio_device.supported_output_configs().expect("error while querying configs");
    for format in supported_configs_range {
        println!("Sample Rate: Min: {:?}, Max: {:?}", format.min_sample_rate(), format.max_sample_rate());
        println!("Channels: {}", format.channels());
        println!("SampleFormat: {:?}", format.sample_format());
    }

    let filename = "rick.mp3";
    let mut fg = Flowgraph::new();

    let src = FileSource::new(filename);
    let inner = src.as_async::<FileSource>().unwrap();

    fn tuple(s: &f32) -> Vec<f32> {
        let mut vec = Vec::with_capacity(2);
        vec.push(*s); // L
        vec.push(*s); // R
        return vec;
    }
    let custom = ApplyIntoIter::<f32, Vec<f32>>::new(tuple);
    let snk = AudioSink::new(inner.sample_rate(), 2); // inner.channels()

    println!("File: {}, Sample Rate: {}, Channels: {}", filename, inner.sample_rate(), inner.channels());

    let src = fg.add_block(src);
    let custom = fg.add_block(custom);
    let snk = fg.add_block(snk);

    fg.connect_stream(src, "out", custom, "in")?;
    fg.connect_stream(custom, "out", snk, "in")?;

    let rt = Runtime::new();
    rt.run(fg).unwrap();

    Ok(())
}
`

Thanks to @loic-fejoz for providing the sample.

loic-fejoz, 2022-04-07:
Indeed. In GNU Radio, you can  specify the number of channels as per our constructor. Yet you will end up with the following block for instance for a stereo AudioSink.
[image]
I could also imagine that, because we might want to keep channels in sync, we could have only one input, either of type `Complex<f32>`, or, if we generalise, of type `IntoIter<f32>` with an iterator always providing at least the number of channels declared.

Maybe we could come up with different constructors enabling FutureSDR to have only one implementation supporting those different types.

But currently, the implementation is the one providing the most surprise...

By the way, I use an implementation using vec<f32> but one could have done it also with `std::iter::repeat` chained with `std::iter::take` as done in the [CW example](https://github.com/FutureSDR/FutureSDR/blob/master/examples/cw/src/lib.rs#L81). Types are just more complicated to find.

lkolbly, 2022-04-11:
I wonder if we could make `StreamIo` have an explicit notion of "wide samples" (in other words, samples which are composed of many sub-samples).

For example, this:
```rust
StreamIoBuilder::new().add_input("in", 4).build()
```
would get replaced with:
```rust
StreamIoBuilder::new().add_wide_input("in", channels, 4).build()
```
and this:
```rust
let i = sio.input(0).slice::<f32>();
// i[idx * num_channels + channel] to get channel `channel` of index `idx`
```
would be replaced by:
```rust
let i: FutureSdrWideSampleSlice<f32> = sio.input(0).wide_slice::<f32>(num_channels);
// i[idx][channel]
```
this would be, in effect, equivalent to using the type `[f32; num_channels]`, but with the difference that it could be run-time dynamic.

If `AudioSink` took a generic type that implements `IntoIter<f32>`, then I think either the user would need to specialize for every channel count they could want, or the concrete type would need to be a `Vec<f32>`, which has certain performance implications.

Notably, `AudioSink` isn't the only block that has a similar problem, the `Fft` block does a similar thing.

I might play around with this for a bit and see how it works out.

bastibl, 2022-04-12:
For multi-channel audio data, there are two possible internal representations: interleaved (for stereo: r0, l0, r1, l1, r2, ...) or deinterleaved (possible in separate buffers, e.g.: r0, r1, r2, ... | l0, l1, l2).

The question is what's more practical. GNU Radio forces you to deinterleave data to put it in two separate input buffers. I don't think that this is a good idea since both audio files and cpal audio sources/sinks use interleaved data. So with GNU Radio you would have to deinterleave samples to put it in two separate stream inputs and inside the audio sink interleave immediately again. This is not optimal. For that reason, I would stick with one input and have a parameter that indicates the number of channels. 

I also would not change the input type. It's fine to stick to single samples (floats or integers) and not try to introduce an artificial grouping. According to my understanding that also what cpal does. Let's say you read interleaved samples from a file and want to forward it to a sink and would be forced to convert it to complex numbers in-between for no reason. So there is no smartness from the runtime involved to make sure your audio data is as expected by the sink. You'd have to make that this is the case in the application/flowgraph.

The idea of the wide samples seems to be similar to GNU Radio's vectors. I also thought about it, but actually I wanted to keep this feature out of the runtime, since it introduces considerable complexity that is not really needed. The FFT, for example, can calculate the available full vectors itself `input.len()/fft_size` and only process those. This keeps the runtime simple and is not a big burden for the block implementation. But if we want something like that, we could maybe look into creating something like a wrapper block that checks for invariants before calling the inner work() function?

loic-fejoz, 2022-04-26:
While I now better understand the rational behind interleaved or not as explained by @bastibl, I believe Rust has some trick to keep this performance while still providing better feedback through the type system.
So I would push to have kind of type "alias" of f32 for interleaved stream or more precisely the [newtype pattern](https://doc.rust-lang.org/stable/book/ch19-04-advanced-types.html#using-the-newtype-pattern-for-type-safety-and-abstraction) so as to have best of both world (performance & type system).

To summarise, I would expect `AudioSink` to require the following type:
```rust
struct InterleavedF32(f32);
```

bastibl, 2022-04-27:
I fully agree. I just have no good idea how exactly it could look. This struct doesn't encode how much streams are interleaved. Considering also FFTs, for example, one might want this. Problem is that, at the moment, the runtime just checks for the correct data size. And if one considers arrays/vectors of items, this would break. Maybe one could implement a `StreamData` trrait that could be implemented for primitive types and interleaved/vector types that implements a method to check whether it is fine to connect. Just a rough idea, though.

loic-fejoz, 2022-04-27:
So maybe with [const generic](https://blog.rust-lang.org/2021/02/26/const-generics-mvp-beta.html), something like:
```rust
struct StreamData<T, const size: usize>(T)
```

bastibl, 2022-04-28:
What I meant was something like

```rust
trait StreamData {
    fn connectable(self, other_endpoint: Box<dyn StreamData>) -> bool;
}
```

But I'm not sure, if this can work. I would also have to try it. The main idea would be that you could use `[f32; 2]` as input type for audio sink to clearly indicate that it is interleaved stereo. (Or even use a separate new type F32Stereo or what ever, which would just be an alias.) Then f32 and [f32; 2] would have to implement the StreamData trait and implement connectable in way that it allows the connection. But there are several issues with this approach, which would complicate the runtime. Not sure, if this is really worth it. At the moment, I don't have a good idea how to do it.

bastibl, 2022-09-16:
I think with the ApplyNM block, we have a pretty decent tool now to work with interleaved, multi-channel samples. I close this, for now. Let's continue the discussion, if we experience any concrete issues with this approach.