FutureSDR/FutureSDR #50: Option to change soapy device properties at runtime

JuliDi, 2022-04-16:
Hey,

Is it possible to change soapy device properties like the center frequency etc. at runtime?  Or more general: Can the flowgraph be changed at runtime?

I would like to avoid stopping the flowgraph to create a new `SoapySource` object, which will cause a new device initialisation that takes a few seconds.
And if so, how could that be implemented in FutureSDR? I did not find a way to access the device after it is converted to a block.

From what I see in the soapy crate doc (https://kevinmehall.net/rustdoc/soapysdr/soapysdr/struct.Device.html#method.frequency) this _might_ be possible, but I am anything but an expert on this.

What I would like to do at the end is an extension of the `fm-receiver` example:

```rust
let src = SoapySourceBuilder::new()
        .freq(93.7e6)
        .sample_rate(3.2e6)
        .gain(40.0)
        .build();

// Connect everything ...

// Execute 
Runtime::new().run(fg)?;
// as an asynchronous task/thread/whatever

// Make changes to the frequency to tune e.g. to a specific radio station on the go
```

bastibl, 2022-04-25:
The Soapy Source already has a basic handler to change the frequency: https://github.com/FutureSDR/FutureSDR/blob/master/src/blocks/soapy_src.rs#L36-L55 This could be extended to accept other types of PMTs. And one could add more handlers in a similar fashion.

Instead of `run`ning the flowgraph, which awaits the task until the flowgraph is finished. You can [`start`](https://docs.rs/futuresdr/latest/futuresdr/runtime/struct.Runtime.html#method.start) it, which gives you a [FlowgraphHandle](https://docs.rs/futuresdr/latest/futuresdr/runtime/struct.FlowgraphHandle.html). With the handle you can call message handlers of blocks. Alternatively, you can use the REST API to issue callbacks.

JuliDi, 2022-04-25:
Thanks for the input, I will give that a try!