FutureSDR/FutureSDR #55: FM Receiver Resampling Issue

bastibl, 2022-06-06:
I was trying to extend the FM receiver example. I added a function to the audio sink that allows querying the default sample rate of the device https://github.com/bastibl/FutureSDR/commit/8da5ceabd1c835dd6b7f9b1277ee1636d4060ca3. For me, this returns 44.1kHz. Like in the current example, I wanted to resample in two steps with an intermediate rate of `5 * audio_rate`. Using this in combination with my HackRF at a sample rate of 8MHz resulted in a rather long filter (~10k taps). With the default buffer size, this stalls the flowgraph, since only 4k samples fit in the buffer. Increasing the buffer size with a `config.toml` works. However, the audio is broken.

My WIP state is here https://github.com/bastibl/FutureSDR (the master branch). @AndersKaloer do you have any idea? I'm not sure if this is a problem with the resampler or my code.

Here's the diff with debug output: https://github.com/bastibl/FutureSDR/commit/6fcb45c931c85825fd25894361df05ce50c8ca7f

bastibl, 2022-06-06:
Might be because of the DC offset, will try offset tuning.

bastibl, 2022-06-07:
Tuning the radio quite a bit away (>0.5MHz) results in some audio. I'm still not sure if this is a bad receiver design of if something is wrong with the resampler. Would be great if you could have a look.

AndersKaloer, 2022-06-07:
I will try to have a closer look at this tonight.

It sounds a bit like the GCD normalizer in `FirBuilder::new_resampling` does not work as intended. The filter should produce `2*12*interp` taps, and in your case I think the GCD is 80000, so effectively you only need to interpolate by 3 and decimate by 100, which should produce 72 taps. Maybe you could try specifying 3 and 100 directly?

I also don't think you want a complex filter here, since that will remove the negative frequencies, which contain half of the signal power in FM.

In any case, downsampling by 100 is still quite much for a filter of that length, and it might be better to use a chain of resamplers, e.g., a filter that interpolates by 3 and decimate by 10 followed by another filter that interpolates by 1 and decimates by 10.

Also, I had to compile with `--release` in order to make it perform well (but I guess you are aware of that).

bastibl, 2022-06-07:
I think the GCD works. It's gcd(44100*5, 8000000) = 500, so it's `2*12*441 = 10584`. My feeling is that there is an issue with larger decim/interp rates, but I didn't have a closer look. It's not urgent, I don't want to annoy you :-)

AndersKaloer, 2022-06-08:
I tried your code, and it worked fine for me both with BladeRF and HackRF, and with 8 MHz bandwidth (the only problem I had was when my bandwidth was smaller than `2*OFFSET`).
I run it using (`PKG_CONFIG_ALLOW_CROSS` and the target flag are just to compile for aarch64-apple-darwin)
```
PKG_CONFIG_ALLOW_CROSS=1 cargo run --target aarch64-apple-darwin -- --frequency 89700000 --rate 8000000
```

I also tried to dump the samples to a file both before and after the `resamp1`.
The input to `resamp1` looks like this:
[image]
The output looks like this:
[image]

Note that my PC actually reports 48000 as the audio rate, which means that I am effectively interpolating by 3 and decimating by 100, so I do not have so many taps in my filter. Actually, the current filter implementation works as intended, but obviously it produces a large number of taps when the interpolation factor is high. Honestly, it is quite inefficient to have such high interpolation/decimation factors in a single filter, so a much better solution would be to adjust the sampling frequency a bit to have a for higher GCD. 

What you can do is to design your own filter with a smaller number of taps. Something like this:
```rust
let gcd = num_integer::gcd(interp, decim);
let interp = interp / gcd;
let decim = decim / gcd;
let resamp1_taps = firdes::kaiser::multirate::<f32>(interp, decim, 2, 0.0001);
println!("{} taps", resamp1_taps.len());
let resamp1 = FirBuilder::new_resampling_with_taps::<Complex32, Complex32, _>(
    interp,
    decim,
    resamp1_taps,
);
```

BTW, be careful with https://github.com/bastibl/FutureSDR/commit/6fcb45c931c85825fd25894361df05ce50c8ca7f#diff-d0c711267097b2b1b136ad18d24dbf53f875886e8f713a08be4b816e8ed7fec0R176-R178 since the filter taps are usually designed for specific values of interp and decim.

bastibl, 2022-06-08:
Thanks for looking into it and for your feedback. I deleted the bogus gcd-minimization and tried to make the receiver smarter selecting the parameters. (Setting the offset to sample_rate / 4, choosing audio_rates based on gcd, automatically selecting the audio multiplier.) I think it's pretty OK now. At some point, one could add dedicated blocks for frequency shifting and demodulation, which should provide better performance compared to the Apply-style blocks.

Thank you very much!

closed 2022-06-08 by bastibl