FutureSDR/FutureSDR #57: Examples with wasm compilation target broken because of missing crate::async_io

j-chmielewski, 2022-06-11:
Ran into this while trying to run examples:

* cw
* wgpu
* spectrum
* wasm

The error:
```
   Compiling futuresdr v0.0.18 (/home/jck/workspace/FutureSDR)                                                                                                                                                     
error[E0433]: failed to resolve: could not find `async_io` in the crate root                                                                                                                                       
  --> src/runtime/mocker.rs:72:16                                                                                                                                                                                  
   |                                                                                                                                                                                                               
72 |         crate::async_io::block_on(async move {                                                                                                                                                                
   |                ^^^^^^^^ could not find `async_io` in the crate root                                                                                                                                           
                                                                                                                                                                                                                   
error[E0425]: cannot find function `block_on` in this scope                                                                                                                                                        
   --> src/runtime/runtime.rs:119:9                                                                                                                                                                                
    |                                                                                                                                                                                                              
119 |         block_on(rx).expect("run_flowgraph did not signal startup completed");                                                                                                                               
    |         ^^^^^^^^ not found in this scope                                                                                                                                                                     
    |                                                                                                                                                                                                              
help: consider importing one of these items                                                                                                                                                                        
    |                                                                                                                                                                                                              
5   | use futures::executor::block_on;                                                                                                                                                                             
    |                                                                                                                                                                                                              
5   | use futures_lite::future::block_on;                                                                                                                                                                          
    |                                                                                                                                                                                                              
5   | use futures_lite::stream::block_on;                                                                                                                                                                          
    |                                                                                                                                                                                                              
                                                                                                                                                                                                                   
Some errors have detailed explanations: E0425, E0433.                                                                                                                                                              
For more information about an error, try `rustc --explain E0425`.                                                                                                                                                  
error: could not compile `futuresdr` due to 2 previous errors                                                                                                                                                      
Error: Compiling your crate to WebAssembly failed                                                                                                                                                                  
Caused by: failed to execute `cargo build`: exited with exit status: 101                                                                                                                                           
  full command: "cargo" "build" "--lib" "--target" "wasm32-unknown-unknown" "--features=wgpu"                                                                                                                      
make: *** [Makefile:4: wasm] Error 1                                                                                                                                                                               
```

I guess it's caused by conditional compilation of async_io re-export while code compiled to wasm still requires it (src/lib.rs):
```rust
// re-exports
#[cfg(not(target_arch = "wasm32"))]
pub extern crate async_io;
```

If you can think of easy enough fix I'd be glad to contribute.

bastibl, 2022-06-12:
Hi, thanks for bringing this up! I noticed it last week but forgot to push the changes to the upstream repo :-/ sorry. Can you try the current main branch and see if it works for you? I hope I find some time to add a WASM CI build soon.

j-chmielewski, 2022-06-12:
Yes, it does work now, thanks :).

closed 2022-06-13 by j-chmielewski