FutureSDR/FutureSDR #58: Two different crates trying to link different versions of the clang native library

jschiefer, 2022-06-21:
When opening the FutureSDR repo with VS Code with the rust-analyzer extension installed, there are two persistent error pop-ups:

[image]

From looking at the logs, this appears to be the result of the rust-analyzer back-end running the command `cargo check --workspace`. Running the same command from the command line shows the problem: 
```
PS C:\Users\jas\Develop\FutureSDR> cargo check --workspace
error: multiple packages link to native library `clang`, but a native library can be linked only once

package `clang-sys v0.21.2`
    ... which satisfies dependency `clang-sys = "^0.21.0"` (locked to 0.21.2) of package `bindgen v0.32.3`
    ... which satisfies dependency `bindgen = "^0.32.3"` (locked to 0.32.3) of package `lttng-ust-generate v0.1.1`
    ... which satisfies dependency `lttng-ust-generate = "^0.1.1"` (locked to 0.1.1) of package `futuresdr v0.0.19 (C:\Users\jas\Develop\FutureSDR)`
    ... which satisfies path dependency `futuresdr` (locked to 0.0.19) of package `androidfs v0.1.0 (C:\Users\jas\Develop\FutureSDR\examples\android)`
links to native library `clang`

package `clang-sys v1.3.3`
    ... which satisfies dependency `clang-sys = "^1"` (locked to 1.3.3) of package `bindgen v0.57.0`
    ... which satisfies dependency `bindgen = "^0.57.0"` (locked to 0.57.0) of package `soapysdr-sys v0.7.3`
    ... which satisfies dependency `soapysdr-sys = "^0.7.3"` (locked to 0.7.3) of package `soapysdr v0.3.2`
    ... which satisfies dependency `soapysdr = "^0.3.1"` (locked to 0.3.2) of package `futuresdr v0.0.19 (C:\Users\jas\Develop\FutureSDR)`
    ... which satisfies path dependency `futuresdr` (locked to 0.0.19) of package `androidfs v0.1.0 (C:\Users\jas\Develop\FutureSDR\examples\android)`
also links to native library `clang`
```
I see this on both Windows 10 and Ubuntu 22.04. It would be nice not to have those errors pop up every time you open FutureSDR with VS Code.

bastibl, 2022-06-24:
Thanks for opening the issue. I took it as the last nudge to finally fix the crate and sub-crate structure of the project. I did quite a bit of rearrangement. This error should no longer occur. Would be great if you could give it a try.

bastibl, 2022-06-24:
And now it's also in the main repo -.-

jschiefer, 2022-06-24:
Thank you for the quick turnaround! Yes, the problem that the issue was about, is fixed. I think this is great, because it avoids people from getting confused/distracted when they open the repo for the first time.

However, there is an undesirable side effect. In the previous version, all the examples would (eventually) automatically get little `Run|Debug` annotations, like so:

[image]

This made it really easy to quickly try out a few examples. With the fix for this issue in place, this no longer happens, presumably because the examples are no longer automatically analyzed. Also, when generating a `launch.json` file automatically, the examples are no longer included. So the net effect is that there is now a lot more hassle involved in running the examples.

bastibl, 2022-06-25:
Hmm, I see... that's a downside of the current approach :-/ But I guess one has to die one death here. Some more context: Before the refactor, all examples and performance evaluation projects were sub-crates of the same workspace. Since there are examples for basically every feature available, commands like `cargo check --workspace` enable all features (i.e., it enables all features that any example/perf sub-crate might enable). This means, if you don't have all (partially obscure dependencies installed), these basic commands would fail. I guess that's the worse case.

With the current project structure, each example is an independent crate that just happens to live in the same directory structure. I don't use the code editor. I assume you'd open the example folder as project, instead of the base directory.

I'm not sure if there are better ways to do it. I'll have a look at some of the popular crates to see how they do it. Apart from that, it should probably be documented that one should open the examples as projects. I put it on my list.

jschiefer, 2022-06-27:
Fair enough, having to have all dependencies for all obscure features seems like an uphill battle. So I would agree that the new refactored way is much preferable.
I tried the approach you suggested (opening examples with VS Code, e.g. `examples/audio`), and this worked fine and seems like a perfectly workable solution. Thank you for taking care of this!

jschiefer, 2022-07-27:
This can probably be closed. Would you like me to do this?

bastibl, 2022-07-28:
I'll close this one but feel free to close yourself when you feel the issue is resolved. Thanks for bringing it up!

closed 2022-07-28 by bastibl