FutureSDR/FutureSDR #60: Stream Tag Keys

samcarey, 2022-07-28:
Is it possible for stream tags to have keys?
In GNU Radio, tags have both keys and values, but I don't see this in FutureSDR.

bastibl, 2022-07-29:
For now, I mainly focused on adding tags and not really how I want those tags to look like. One could just adapt this enum as needed: https://github.com/FutureSDR/FutureSDR/blob/main/src/runtime/tag.rs#L6-L10 Or maybe make it a struct with lot's of optional fields. I didn't really think about how these tags should look like. Suggestions are very welcome.

bastibl, 2022-09-16:
I implemented more tag types when implementing the WLAN transceiver. https://github.com/FutureSDR/FutureSDR/blob/7238a265ab41ac0384041c66e0b25c7251b70a7a/src/runtime/tag.rs#L39-L48 I have the feeling that, in the long run, only the `Named...` will turn out to be useful. Let's see. Also, since the enum is `non_exhaustive` we can extend with minimal hassle.

I close this, for now. Feel free to open another issue, when you miss something.

closed 2022-09-16 by bastibl