FutureSDR/FutureSDR #62: Only first TX/RX port can be used on soapy+usrp b210?

astrojhgu, 2022-08-20:
https://github.com/FutureSDR/FutureSDR/blob/77f1a44f59847be85593eed3900b0294e44ea3eb/src/blocks/soapy_src.rs#L146

When getting stream from `soapydev`, the channel is hardcoded to 0.

bastibl, 2022-08-20:
At this stage, the project is unfortunately full of those limitations. I didn't have the time to create a full-fledged block library but usually extend blocks as needed. Any help is greatly appreciated. Feel free to ask questions here or on Discord, if you want to give it a shot.

etschneider, 2022-10-06:
I'm working on multi-channel support. It's currently working, but needs a little more polish. If you'd like to try it out, see:

https://github.com/etschneider/FutureSDR/tree/ets_soapy_multichan

commit 8d154520 (2022-10-08): soapy: add multi-channel support

The implementation should be backwards compatible
via the builder interface. The new() signature changes
the chan param to a Vec.

This builds upon the optional params PR #71.

Fixes: #62
Refs: #70, #71
closed 2022-10-08 by bastibl