FutureSDR/FutureSDR #67: Checking type size is not enough

bytesnake, 2022-10-04:
There is currently a type size check at runtime, preventing out of array access for input consumers. https://docs.rs/futuresdr/latest/src/futuresdr/runtime/topology.rs.html#199 

This does not prevent undefined behaviour, for example it allows to connect `f64` to `Complex<f32>` arguments:

```rust
use futuresdr::anyhow::Result;
use futuresdr::blocks::NullSink;
use futuresdr::runtime::Flowgraph;
use futuresdr::runtime::{Block, Runtime};
use futuresdr::blocks;
use futuresdr::blocks::Fft;

/// Create sine tone.
pub struct Oscillator64;

impl Oscillator64 {
    pub fn new(freq: f64, amp: f64, sample_rate: f64) -> Block {
        let mut arg = 0.0;
        let diff = 2.0 * std::f64::consts::PI * freq / sample_rate;
        blocks::Source::new(move || {
            let s = amp * f64::sin(arg);
            arg += diff;
            arg = arg.rem_euclid(std::f64::consts::PI * 2.0);
            s
        })
    }
}

fn main() -> Result<()> {
    let mut fg = Flowgraph::new();

    let src = Oscillator64::new(440.0, 0.3, 48000.0);
    let fft = Fft::new(512);
    let snk = NullSink::<f64>::new();

    // uh oh \o/
    let src = fg.add_block(src);
    let fft = fg.add_block(fft);
    let snk = fg.add_block(snk);

    fg.connect_stream(src, "out", fft, "in")?;
    fg.connect_stream(fft, "out", snk, "in")?;

    Runtime::new().run(fg)?;

    Ok(())
}
```

bytesnake, 2022-10-04:
no I just started to read the source and stumbled on https://github.com/FutureSDR/FutureSDR/blob/main/src/blocks/fft.rs#L104 which made me take note here. It would be great to at least check that all downstream node's scalar are compatible to the input scalar of current node.

bastibl, 2022-10-04:
You can cast the stream inputs and outputs to anything. That's not specific to this mutable reference to the input. It's the case for all stream connections.

A runtime check should be rather easy to implement by requiring any item type to implement `Any` and store the [TypeId](https://doc.rust-lang.org/std/any/trait.Any.html#tymethod.type_id) in the stream port. But that's just a runtime check... not sure if this is good enough for you.

bytesnake, 2022-10-04:
> A runtime check should be rather easy to implement by requiring any item type to implement Any and store the [TypeId](https://doc.rust-lang.org/std/any/trait.Any.html#tymethod.type_id) in the stream port. But that's just a runtime check... not sure if this is good enough for you.

that would definitely be the reasonable way to go :+1:

bastibl, 2022-10-08:
Implemented in https://github.com/FutureSDR/FutureSDR/commit/ba57854a30b06b9bc6c64e4bea35ccee05b9508c.

Stream inputs/outputs are created with the actual type `add_input::<T>("in")`/`add_output::<T>("out")`. The actual ports remain non-generic, since only the `any::TypeId` of <T> is stored.

This `TypeId` is checked during connection and when `cast`ing the slice in work. There are also `slice_unchecked` versions, which ignore this check, since some blocks are, for example, easier to implement by casting it to a `u8` slice in `work`.

closed 2022-10-08 by bastibl