FutureSDR/FutureSDR #68: Use enumeration to distinguish between samples and messages

bytesnake, 2022-10-04:
FutureSDR is heavily inspired by Gnuradio; For example it distinguishes between messages and data in separate input/output operations [StreamIO](https://docs.rs/futuresdr/latest/futuresdr/runtime/stream_io/struct.StreamIo.html) and [MessageIO](https://docs.rs/futuresdr/latest/futuresdr/runtime/message_io/struct.MessageIo.html) and duplicates blocks respectively (e.g. Copy vs MessageCopy, Sink vs. MessageSink etc.) Further tags have no type associated, restricting it to primitive types or rely on polymophism via type erasing.
This makes working with messages more awkward and implementing combinatiorics on them more difficult.

For me he problem feels very similar to error handling in Rust and has probably a similar, more concrete solution. We could for example bound the input type of a node to implement the conversion traits (e.g. [From](https://doc.rust-lang.org/std/convert/trait.From.html) or [TryFrom](https://doc.rust-lang.org/std/convert/trait.TryFrom.html)) and either have a global enumeration of all possible messages (+ samples) or inter node specific messages. If a downstream library defines their own blocks and messages, they have to implement conversion into futuresdr message type (I'm using messages here for samples as well as data).

I'm not familiar with Gnuradio and not informed about scheduling differences between messages and samples, if there are some the idea may not be feasible.

bastibl, 2022-10-04:
I'd say that FutureSDR is a SDR runtime. And every runtime I know of distinguishes these two primitives. It's just natural for the problem. These two port/connection types also implement very different primitives. One is async message passing and one is high-throughput stream data. It has to me mapped to different underlying mechanism, like channels for messages and some kind of ring buffers (or similar) for streams. So the distinction has to made in the runtime. I'm not sure what exactly you are suggesting in the first place, but you cannot just have one primitive and every sample is an enum of type sample that is pushed through a shared channel or similar.

Of course, this doesn't mean that there cannot be some kind of alignment or convergence on higher layers. But neither is there an obvious API (which would have to cover both usage patters) nor can I imagine a good reason to do it in the first place. Can you maybe provide an example why this should be done?

I know, since this is Rust, one can just say that everything should be put in the type system and statically checked at compile time. But this is not always possible or very hard to do. From your suggestions it is not clear, how exactly this should look. Maybe you could just provide a draft PR or some concrete code example (e.g., how the API is vs how you think it should be), since I don't really understand your issue.

bytesnake, 2022-10-04:
> One is async message passing and one is high-throughput stream data

are the sample based methods using the polling infrastructure? I thought they use the same underlying async runtime only storage is different

> I'm not sure what exactly you are suggesting in the first place, but you cannot just have one primitive and every sample is an enum of type sample that is pushed through a shared channel or similar.

sure but you could have an item in your future indicating whether a new sample happened, a message or both and return a view to your ringbuffer or message

>  [...] nor can I imagine a good reason to do it in the first place

good reasons are simplicity in the API surface, type checked message structs and flexibility for the user what get passed where

hope to have some prototype to not just dumping ideas here :grin:

bastibl, 2022-10-04:
Great, looking forward to your prototypes!

closed 2022-11-05 by bastibl