FutureSDR/FutureSDR #69: Add combinators for blocks

bytesnake, 2022-10-04:
It would be great to have combinators (similar to Iterator's `map`, `filter`, `join`, `reduce` etc.) available for all blocks doing sample or message processing. Normally they are implemented using a trait similar to https://doc.rust-lang.org/std/iter/trait.Iterator.html

bastibl, 2022-10-04:
We already have functional-style blocks (`map` -> `Apply`, `filter` -> `Filter`, etc). See the top [here](https://docs.rs/futuresdr/latest/futuresdr/blocks/index.html). But, at the same time, an SDR is not an Iterator and it is not immediately clear how this could be done. Any concrete suggestions? PRs are very welcome!

bytesnake, 2022-10-04:
to give a concrete example, you can compose types implementing a trait (for async you have the [Stream](https://docs.rs/futures/latest/futures/prelude/stream/trait.StreamExt.html#method.chain) trait as example):

```rust
let block = FileSource::<Complex<f32>>::new("my_filename.cf32", false)
    .map(|sample| sample * 0.5)
    .filter(|sample| sample.abs() < 0.5)
    .join(second_file);

let source = fg.add_block(block);
```

bastibl, 2022-10-04:
At the moment, this would be something like:

```rust
let block = FileSource::<Complex32>::new("my_filename.cf32", false);
let map = Apply::new(|s: &Complex32| s * 0.5);
let filter = Filter::new(|s: &Complex32| if s.abs() < 0.5 { Some(*s) } else { None });

connect!(fg, block > map > filter > second_file);
```
I think this is fine, since the whole point is really to have multiple blocks. At least, this is what has to happen under the hood. Not sure, if it is desired to fake this abstraction. If you want to chain iterators within one block, this could be done with the [Iterator block](https://docs.rs/futuresdr/latest/futuresdr/blocks/struct.ApplyIntoIter.html). Then you have file source -> chained iterators in one iterator block -> file sink

closed 2022-11-05 by bastibl