FutureSDR/FutureSDR #77: Allow custom routes to call handlers in blocks

ratzrattillo, 2022-10-09:
With the introduction of `flowgraph.set_custom_routes(router);` we have the option to add custom routes to the axum server backing the FutureSDR control port.
While this is nice we are still having some limitations, as within these custom routes we cannot call the handlers of a block directly.
Example:
The default handler to set the RX frequency of a SoapySource Block to a out of range value might crash the flowgraph. Thus, users might want to introduce bounds checking. Bounds checking can be done in the handler of a custom route, but then how to call the original block handler to set the frequency?

The current workaround is to create a second axum server which can receive the handle to a flowgraph `(let (task_handle, fg_handle) = block_on(runtime.start(fg));` and then giving the fg_handle to the custom route in the scond axum server. This is not very elegant.

Another workaround would be to do the bounds checking in a custom route and then create a new HTTP request to the handler function in the SoupySource Block. Not too elegant either.

We thus need a way to allow custom routes to efficiently call handler fucntions of a block in a flowgraph.
Also, we should considering that the axum server is not bound to a flowgraph, but rather to the runtime, meaning having not one axum server per flowgraph, but one per runtime instance.

etschneider, 2022-10-10:
You can call handlers in custom blocks, but you have to be a little tricky to get access to the `FlowgraphHandle`.

Create an `Arc<Mutex<Option<FlowgraphHandle>>>` that you give to your router and set it with the value returned from `runtime.start()`.

Try something like this:
```rust
fn main() -> Result<()> {
    let mut fg = Flowgraph::new();
    let arc_opt_fg_handle: Arc<Mutex<Option<FlowgraphHandle>>> = Arc::new(Mutex::new(None));

    fg.add_block(
        MessageSourceBuilder::new(
            Pmt::String("foo".to_string()),
            time::Duration::from_millis(100),
        )
        .build(),
    );

    let router = Router::new()
        .route("/my_route/", get(my_route))
        .layer(Extension(arc_opt_fg_handle.clone()));

    fg.set_custom_routes(router);

    println!("Visit http://127.0.0.1:1337/my_route/");
    let (task_handle, fgh) = block_on(Runtime::new().start(fg));
    {
        let mut opt_fg_handle = arc_opt_fg_handle.lock().unwrap();
        *opt_fg_handle = Some(fgh.clone());
    }
    block_on(task_handle)?;

    Ok(())
}

async fn my_route(
    Extension(arc_opt_fgh): Extension<Arc<Mutex<Option<FlowgraphHandle>>>>,
) -> Html<String> {
    //Note: need to clone out of mutext guard, which is not `Send`, which
    //breaks handler due to use of .await (Axum thing)
    let opt_fgh = arc_opt_fgh.lock().unwrap().clone();

    let dstr = if let Some(mut fgh) = opt_fgh {
        format!("{:#?}", fgh.description().await.unwrap())
    } else {
        "".to_owned()
    };

    Html(format!(
        r#"
    <html>
        <head>
            <meta charset='utf-8' />
            <title>FutureSDR</title>
        </head>
        <body>
            <h1>My Custom Route</h1>
            Flowgraph description: 
            <pre>{}</pre>
        </body>
    </html>
    "#,
        dstr
    ))
}
```

etschneider, 2022-10-10:
Also, you mention SoapySource, so a couple points:

- `soapysdr::Device` is cloneable, so you can pass it to a route handler too.
- I would consider the block crashing due to a bad freq a bug in the block which should be fixed.

ratzrattillo, 2022-10-11:
Thanks for the explaination and the solution.
For me it would be nicer, if FutureSDR provided an easy and intuitive way on how to work with the flowgraph handle in custom routes. While your solution is clearly working i do not think that it is very intuitive (Not your fault of course, as you have to work around the current limitations).
I will close this issue nonetheless.
A new issue will be openend to adress the decoupling of axum and the flowgraph and rather coupling it with the runtime.

closed 2022-10-11 by ratzrattillo