FutureSDR/FutureSDR #80: Do not return an error when flowgraph was terminated exteranlly.

bastibl, 2022-10-10:
There is currently no differentiation between termination because of a block error and external termination of the flowgraph through the flowgraph handle. In both cases the flowgraph task returns an `Err("Flowgraph was terminated")`. While this is fine when a block failed, we should probably try to shutdown gracefully in case of external termination. Maybe we should introduce `Kill` (in case of block error) and `Shutdown` (from external). In the latter case, the flowgraph task should succeed. See this example: https://github.com/samcarey/drop_block_test/blob/main/src/main.rs#L20

etschneider, 2022-10-11:
I've been looking into this. The direct fix is easy, just remove the `bail!` in `run_flowgraph`/`FlowgraphMessage::Terminate`, all the blocks get a chance to terminate and the loop exits on the `active_blocks == 0` condition. That seems like the right behavior.

Perhaps there should be a a time limit to wait on the blocks to terminate (in case one fails to terminate correctly).

However, the provided example has different problem. The provided code deadlocks, but not because of the runtime result. 

Interestingly, the `tests/runtime::fg_terminate` does not deadlock, even when checking the `task.await` result. The workaround is simple, but not intuitive (at least to me).

This deadlocks:
```rust
let (task, mut fg_handle) = block_on(Runtime::new().start(fg));
```

This works:
```rust
let rt = Runtime::new();
let (task, mut fg_handle) = block_on(rt.start(fg));
```

samcarey, 2022-10-11:
@etschneider's workaround seems to fix the deadlock for me, and the `deinit` and `drop` functions for the block get called too. Thanks! 

I decided to go ahead and wrap the task and handle in a struct that terminates from it's `drop` method so that cleanup will happen automatically, even if there's a panic. So I think my use case is addressed now!

```rust
struct RunningFlowgraph {
    task: Option<Task<Result<Flowgraph>>>,
    handle: FlowgraphHandle,
}

impl From<(Task<Result<Flowgraph>>, FlowgraphHandle)> for RunningFlowgraph {
    fn from((task, handle): (Task<Result<Flowgraph>>, FlowgraphHandle)) -> Self {
        Self {
            task: Some(task),
            handle,
        }
    }
}

impl Drop for RunningFlowgraph {
    fn drop(&mut self) {
        block_on(self.handle.terminate()).unwrap();
        block_on(self.task.take().unwrap()).err();
    }
}

fn main() {
    let mut fg = Flowgraph::new();
    fg.add_block(ExampleBlock.into());
    let rt = Runtime::new();
    let running_flowgraph = RunningFlowgraph::from(block_on(rt.start(fg)));
}
```

etschneider, 2022-10-11:
@samcarey, does the struct wrapping help in the deadlock code case? (w/o the separate `Runtime::new()`)

AFAIK, drops are always called, even for panics. The exceptions are pretty special cases, a deadlock being one of them (since all execution is stuck and the stack cannot be unwound).

samcarey, 2022-10-11:
No. Even with the struct wrapper, it deadlocks if the Runtime doesn't have its own let binding.
I don't think that's a concern for me, now that I know to create it separately, but it does seem like a pitfall.

I seem to recall other times while writing Rust that I've gotten a "temporary value dropped" error that told me that some variable needed its own let binding before it could be used on a later line. I don't really understand why the Runtime needs its own binding to not deadlock, but I wonder if there's a way to adjust the API so the user knows to do this (or is forced to).

etschneider, 2022-10-13:
@bastibl, what is the desired behavior when a block fails initialization?

I think there is a good case to be made that the runtime should shutdown as gracefully as possible, sending terminate messages to whatever blocks have already initialized. That way they can cleanup any open handles or whatever.

If returning an error, I do think that it should be a logically distinct error (not just a message string). I think #23 is a good idea in that regard.

Forward thinking here, but IIRC there was some talk about wanting to stop a flowgraph and reconfigure it. Do you see that as a complete teardown and rebuild of the flowgraph?

I can imagine a case that it would be nice to keep the scheduler running while making changes to the flowgraph. Like pausing and reconfiguring or swapping out some blocks while the rest runs. I know, I'm getting carried away here...

etschneider, 2022-10-14:
FYI, the deadlock is caused by a temporary `Runtime` variable being dropped on the same line.

Some equivalent code to make it more obvious:

```rust
    // let (task, mut fg_handle) = block_on(Runtime::new().start(fg));
    let task;
    let mut fg_handle;
    {
        let rt = Runtime::new();
        (task, fg_handle) = block_on(rt.start(fg));
    }
```

bastibl, 2022-12-02:
The normal shutdown, was fixed (though not heavily tested). Leaving the panic behavior from the other issue as todo.

closed 2022-12-02 by bastibl