FutureSDR/FutureSDR #85: Soapy: multiple channel configuration

etschneider, 2022-10-17:
I've made a first attempt at:
https://github.com/etschneider/FutureSDR/tree/ets_new_soapy

If anyone interested, please review and comment. This isn't pull ready,I just want to see if this general approach is acceptable before spending more time on it.

I made a new `blocks::soapy` sub-module, since the code is more complex. I'm using the `examples/soapy_multichan.rs` as a test case. (Perhaps some `#[ignore]` tests in the  soapy module would be better? Or both?)

A few points:
- The new 'cmd' port is used for all the new configuration methods, and supports the dict format of the GNU Radio Soapy block, among others.
- The `SoapyMultiConfig` is meant to be used to allow arbitrarily configurations of all channels/directions with a single message
- There is also a new `SoapyCommand` enum, but it doesn't really add anything currently.
- Since the configuration logic is the same for `SoapySource` and `SoapySink`, I've made a generic `SoapyDevice<stream type>` parent for them both. Trying to keep it DRY.
- The `SoapyInitConfig` struct and the Sink/Source builder pattern are at odds... Probably some discussion needed on this point.
- I tend to think that the old `freq` and `sample_rate` ports should be removed. This would be a breaking change, so???

etschneider, 2022-10-18:
I'm not very happy with the current config structure, basically a struct full of options for every setting:

```rust
pub struct SoapyConfig {
    pub chan: Option<usize>,
    pub dir: SoapyConfigDir,
    pub sample_rate: Option<f64>,
    pub freq: Option<f64>,
    pub gain: Option<f64>,
    ...
```
Seems pretty wasteful, especially assuming that the configuration options will grow to support more of the Soapy device API.

I'll play around with something like:

```rust
enum SoapyConfigItem {
    Gain(f64),    
    Freq(f64),
    ...

struct SoapyConfig(Vec<SoapyConfigItem>);
```

The challenge is dealing with the channel number and direction options in a flexible and elegant way. Perhaps just a chan/dir enum item that applies to all subsequent items in the vec.

bastibl, 2022-10-27:
The message handler for the dict is very cool! Maybe it's the way to go and we should remove the other handlers. I had a look at the Soapy API. It's pretty big. I think we should have a rough plan about the way we want to proceed. At the moment, we add stuff as needed, but I don't have a good idea about a more complete implementation. Do you have any thoughts? Should we support the API calls through the dict?
We don't have to implement everything right now, I just want to avoid restructuring the module every time we add a new feature :-)

etschneider, 2022-10-28:
Lots of thoughts... :-)

First, a single `cmd` port is the way to go. Because it accepts a `Pmt`, our handling can extended without breaking the interface. We can even support multiple "flavors" based on the pmt contents.

The `Vec<SoapyConfigItem>` form is far more powerful IMO. I'm pretty sure it could handle arbitrarily complex scenarios with anything in the soapy API without introducing breaking changes.

However, without `Pmt::Any` serialization it becomes less accessible. I think the next best would be to expose `Vec<SoapyConfigItem>` as a `Pmt::VecAny`. A new convenience type like `Pmt::Named(String,Pmt)` would also help with this, e..g `Pmt::VecAny([Pmt::Named("setting_type",setting_val),..])`.

The current "dict" interface should be kept as the easy/default flavor. (We can add more supported keys as needed)

I'm also in favor of adding an optional user defined `cmd` handler for those one-off "arbitrarily complex scenarios". That way we can keep the logic in the Soapy blocks trimmed down to typical use cases while not inhibiting users that need more. Clearly such a callback would not be serializable for remote instantiation, but still useful.

etschneider, 2022-10-28:
As an example of an odd Soapy use case, I sometimes use the USRP B210 to process two low rate signals at moderately separate frequencies. The B210 does not have independent LOs for each channel, so I do this by setting the LO between the two frequencies of interest and use the DDCs for each channel offset. To do this requires knowing the frequencies of both channels at the time of the tuning command, something that is not possible with a single dict (without adding keys like "freq1", "freq2"). Plus the vec form provides a means of specifying the order of operations.

There are also cases where one would want to update the tx and rx settings at the same time, thus the reason for all the `SoapyDirection` stuff. A sink "cmd" port can alter the source settings and vice versa (it is a single device after all).

BTW, my next steps would be to add timed commands and stream tag support.

bastibl, 2022-11-05:
I didn't have a detailed look yet, but is it possible to make the config and command stuff an expert feature that does not complicate simple use-cases, i.e., the builders construct them internally and the normal user doesn't see them or care about them?

If the new API is purely optional, I think we could just give this a try. Otherwise, every rough edge would be part of any SDR hello world example.

Regarding the removal of the existing handlers, maybe that was not the best idea. Maybe the more complex commands/configs could use message handlers internally that could also be called individually (like the `set_freq`, `set_gain` handlers now). Same for the _dict approach_ that should have keys, corresponding to individual handlers (`set_freq`, `set_gain`, ...). I guess that would keep the typical use-case simpler, as it is not necessary to always create an additional dict that wraps the call.

What did you consider for the `cmd` method for the "arbitrary complex scenarios"? A closure that is passed to the block, gets the device as parameter, and is executed in the context of the block task?

etschneider, 2022-11-06:
> I didn't have a detailed look yet, but is it possible to make the config and command stuff an expert feature that does not complicate simple use-cases, i.e., the builders construct them internally and the normal user doesn't see them or care about them?

The current implementation is backwards compatible, including ports and builders. (With some caveats for multiple channel configuration)

> If the new API is purely optional, I think we could just give this a try. Otherwise, every rough edge would be part of any SDR hello world example.

It is. I'll do a little cleanup and start a PR.

> Regarding the removal of the existing handlers, maybe that was not the best idea. Maybe the more complex commands/configs could use message handlers internally that could also be called individually [...]

The current (backwards compatible) ports are `freq` and `sample_rate` (which is odd to me). If you want to keep the single purpose ports for ease of use, then I'd limit them to `freq` and `gain` (which would break anything currently using `sample_rate`).

These separate ports can only operate on a single channel though.

I lean toward removing them completely (later) and only having the `cmd` port, but I don't feel strongly about it. Helper functions could be created to avoid the little extra typing needed to create a cmd PMT.

> What did you consider for the `cmd` method for the "arbitrary complex scenarios"? A closure that is passed to the block, gets the device as parameter, and is executed in the context of the block task?

Yep. The dev and the `cmd` PMT as closure parameters. Or maybe the full block instead of the just the dev.

etschneider, 2022-11-08:
I just added PR #91. I removed the `sample_rate` port since it didn't seem to be used by anything. So there are three ports: "freq", "gain", "cmd".

The custom "cmd" handler is not included. I have another branch with it, but it needs more testing. I'll do another PR for that sometime later. I'm guessing that no one is in a hurry for that anyway.

PR #91 (2022-11-10): soapy: multi-channel configuration.
Supports individual channel configuration via builder and runtime msg ports.
Fixes: #85
commit 9e87fe81 (2022-11-10): soapy: multi-channel configuration.
Supports individual channel configuration via builder and runtime msg ports.
Fixes: #85
closed 2022-11-10 by bastibl