FutureSDR/FutureSDR #87: macOS support? (making lttng optional)

phstrauss, 2022-10-21:
Hello,

I'm new to FutureSdr and Rust, great project btw, and I'm stuck on macOS for some time, is it not too difficult to makes the lttng dependency optional as it's heavily linux dependent? Otherwise, what needs to be done to build and run on macOS?

Regards.

bastibl, 2022-10-21:
The lttng tracepoints are optional and not enabled by default. So I'm not sure I understand the issue. There shouldn't be anything specific to FutureSDR. Just `cargo build` the project or one of the examples.

phstrauss, 2022-10-21:
Oh sorry, I didn't tried cargo build but stopped on the failure of the check.sh script. You can close the issue. Sorry for the noise.

closed 2022-10-21 by bastibl