FutureSDR/FutureSDR #89: Look for a way to shutdown gracefully when a task/block panics

bastibl, 2022-11-05:


etschneider, 2022-11-09:
I think the first priority is to not to panic. :-)

There are a lot of unchecked results that unnecessarily end up as panics.

After that, we can use [catch_unwind](https://doc.rust-lang.org/std/panic/fn.catch_unwind.html) as an *imperfect* last effort to catch whatever else we can. But not for targets that abort rather than unroll, like wasm ([link](https://users.rust-lang.org/t/catch-panic-in-wasm/57569)).

As a first step, I recommend making `run_flowgraph` and `run_block` not return `Result`, as they are thread/task top level routines and if they can't catch their errors then the runtime is pretty borked anyway and a panic is the reasonable action. Not to mention that both methods take ownership of their flowgraph/block which will then be dropped on `Error`.

After that we can work through the rest of the runtime adding error handling.

This is pretty closely related to #80/#81. I'll push the `run_flowgraph` modification to #81 for review.

etschneider, 2022-11-09:
So changing the signature of `run_flowgraph` requires changes to a lot of consumers (examples, tests, etc). Shocking, lol...

Prior to making breaking changes, I think a more wide ranging architecture discussion should take place. At least that way we can try to break it only once.

bastibl, 2022-11-09:
I agree with the goal to not panic, but it should be handled better :-) Even if it's only handy during development of new blocks.

I quickly tried catch_unwind before opening the issue, but it was missing a trait bound. It can probably be fixed, but I didn't have the time at that moment.

I don't get the issue with the return values, though. If an SDR disconnects, or the network goes down, or something runs out of memory, of a file sink notices that the disk is full. These are all errors that should be propagated. The runtime would be borked, if it would panic in that case. But propagating the error is perfectly fine imho. Otherwise, one gets a "successfully" executed flowgraph back and wonder, for example, why the file sink doesn't contain the expected data. And since neither the block or the flowgraph can handle the error, they propagate it and the developer can decide. If dropping a erroneous block or flowgraph is fine, might not be clear. But these two functions could get custom error types that contain the broken block/flowgraph.

etschneider, 2022-11-09:
> I agree with the goal to not panic, but it should be handled better :-) Even if it's only handy during development of new blocks.

I don't think it is possible. Panic is the last resort.

> I don't get the issue with the return values, though. 

It's not really the `Result` return signature that is the problem, it is that there is no way to keep the flowgraph in existence on error. It is moved into the function and only returned on `Ok`, so it is dropped if `Err` is returned. That doesn't seem desirable.

Same thing with `run_block` taking ownership of its `Block`.

So, I recant my suggestion of not returning `Result`, but there needs to be a way to not drop the flowgraph and blocks on errors.

bastibl, 2022-11-10:
I don't get your reply regarding panic. What is not possible. Doing something else than hanging the application on panic?

And regarding the result. As I said, you can return the flowgraph or block also in the error case, using a custom error type that contains an indication what went wrong and the flowgraph.

etschneider, 2022-11-10:
> I don't get your reply regarding panic. What is not possible. Doing something else than hanging the application on panic?

I mean a general way to catch panics, like C++ exceptions.

I'm quite sure that the hanging issue is solvable. 

I'll code up some test cases, I don't seem to be communicating very clearly.

> And regarding the result. As I said, you can return the flowgraph or block also in the error case, using a custom error type that contains an indication what went wrong and the flowgraph.

Passing the flowgraph back in a custom error just to keep it alive? Then it is the responsibility of all the callers to extract the fg from the error and keep it around (all the way up the error chain). 

I'm using `run_flowgraph` as an example, but there is the same issue with `run_block`. Probably more so.

What about just _not  moving_ into the tasks? For example, hold the structs at a higher level in an `Arc<Mutex<>>`, which you then pass to the tasks. The `run_flowgraph`/`run_block` just holds the lock until it exits.

bastibl, 2022-11-10:
And then users outside the flowgraph can compete for the lock? And call arbitrary methods? And the runtime has to get the lock all the time, i.e., for every interaction with the block? That's what GNU Radio is doing and that didn't turn out well. I don't get your issue with the current design...

etschneider, 2022-11-10:
> And then users outside the flowgraph can compete for the lock? And call arbitrary methods? And the runtime has to get the lock all the time, i.e., for every interaction with the block? That's what GNU Radio is doing and that didn't turn out well. I don't get your issue with the current design...

No, no, and no. :-) 

I'm not proposing anything I'd consider a major change to the architecture.

My issue is that I think the flowgraph and block structs need to have longer lifetimes their "run" tasks. That is all.

The arc/mutex is simply an easy way to get a reference to the longer lifetime flowgraph/block struct into the task. There is no expectation that any end user would ever see this. The system _may_ acquire a lock during setup and teardown if needed, but never during runtime.

etschneider, 2022-11-19:
I pushed an example of the Runtime::Arc<Mutex<Flowgraph>> to PR #81, in case you are interested. The changes are pretty minor. Most of the diff is due to indentation (using reference instead of taking topology in `run_flowgraph`).

https://github.com/etschneider/FutureSDR/blob/fg_terminate_no_err/src/runtime/runtime.rs

bastibl, 2022-11-21:
I don't get your point. The flowgraph lives long enough. It is transferred to the executing task and then comes back. I don't see how the commit makes things better. It looks like it only hides the actual design through this `Arc<Mutex<>>` construct. What's the benefit?

I think, the problem is exactly the opposite: It's not that the flowgraph does not live long enough but the runtime is destroyed to early.

In the case of this:
```rust
let (task, mut fg_handle) = block_on(Runtime::new().start(fg));
```
vs this:
```rust
let rt = Runtime::new();
let (task, mut fg_handle) = block_on(rt.start(fg));
```
The issue is that, in the former case, the Runtime is immediately destroyed after calling `start()`. And this is just plain wrong. No `Arc` fixes this.

What's missing is a proper lifetime annotation that would allow the compiler to error out if the runtime/scheduler doesn't live long enough. It would ensure that the lifetime of the runtime matches/exceeds the lifetime of the async closure/task that is spawned on the runtime. It would look like this:
https://docs.rs/smol/latest/smol/struct.Executor.html#method.spawn

But this is very hard to do. If you figure it out -> instant merge :-)

etschneider, 2022-11-21:
The flowgraph only comes back w/ Ok(fg), not on error. 

The modification keeps the fg (and topology) alive on error. The `Arc<Mutex>>` simply provides a long lived owner that can be used in another thread (Send+Sync).

Regarding `Runtime` lifetime, I agree, but that is another (related) issue.  I mentioned that *exact* point in #80.

I'm pretty sure that fix will involve changing the top level runtime API, which I tried to avoid for this small example of what I was talking about w/ the `Arc<Mutex<>>`  (not continuous lock/unlock or external access to fg).

I think that `Runtime::start()` should not return the task (or flowgraph), but rather `Runtime` should own them internally. Maybe with the API something like:

```rust
pub async fn start(&self, fg: Flowgraph) -> Result<FlowgraphHandle>
pub fn wait_done(&self) -> Result<()>
pub fn run(&self, fg: Flowgraph) -> Result<()> 
// If desired
pub fn take_flowgraph(&self) -> Result<Flowgraph>
```

I'd probably make a few other minor changes, but that is the gist.

bastibl, 2022-12-05:
Added `catch_unwind` to worker threads to terminate application and not deadlock: https://github.com/FutureSDR/FutureSDR/commit/0378d8666f072dc009bff29c5336250705b981fc

closed 2022-12-05 by bastibl