FutureSDR/FutureSDR #104: VCO and similar blocks

loic-fejoz, 2022-12-30:
Voltage Controlled Oscillator (VCO) and other similar blocks are missing.
I believe we can come with a common implementation for various use-cases.
At least propose implementation for:

- VCO taking as input the angle rate as in GNU Radio https://wiki.gnuradio.org/index.php/VCO_(complex)
- CO taking as input the frequency

It should also be a base for Phase Shift Keying (PSK, eg BPSK, GMSK, QPSK) and other [constellations](https://wiki.gnuradio.org/index.php?title=Constellation_Object) handling.

Implementation experimentation started in my branch: https://github.com/loic-fejoz/FutureSDR/blob/feature/vco/src/blocks/signal_source/controlled_oscillator.rs

bastibl, 2023-01-08:
Would be really nice to have. I just wonder how we should proceed in the future. Add this in-tree? Add this as an example that comes with some blocks? Have it as separate crate (for example maintained by you)? Maybe, we have examples from the FutureSDR project and 3rd party ones; and all are just git submodules in the example directory?

I'd say it's clear that a new users should find a PSK example, either as example in-tree, a git submodule, or a readme in the examples directory with a list of pointers to external examples.

Just trying to balance "batteries included" vs maintainability. Do you have any input/ideas?

loic-fejoz, 2023-01-08:
I was also having the exact same question.

I believe the underlying block should be included because it is a basis for a lot of graph. Yet I am trying to build one that is very generic and not tied to a particular scheme (modulation), said otherwise, everything consuming a stream and using a numerically controlled oscillator.
So I am shifting the question to the builders: should the builders (for different modulations) be part of FutureSDR or external crates? And the more I think, the more I think it should be part of another crate/project.

Somehow, I would defined the threshold based on technical difficulties of correct implementation. The difficulty to come up with a correct NCO is a one-order higher than defining modulation...

But this is just my 2 cents. What do you think?

loic-fejoz, 2023-03-19:
Should be "closed as not planned" as it will be handle by pull request https://github.com/FutureSDR/fsdr-blocks/pull/4 and issue https://github.com/FutureSDR/fsdr-blocks/issues/2

closed 2023-03-19 by loic-fejoz