FutureSDR/FutureSDR #114: OpenTelemetry Integration

ratzrattillo, 2023-05-24:
# Problem
When designing a flowgraph, various blocks have to play together well in order to encode or decode a signal.

However, we are often blind as soon as the flowgraph is running and use ugly println debugging to know what is going on in a block.
So, while we can clearly look at the source and try to understand what they are doing, we do not have an insight on the concrete data/samples they operate on over time.

# Gaining Observability
To tackle that lack of observability, projects often use tools like OpenTelemetry to record logs, traces and metrics.
The questions that telemetry might answer are e.g.:
- Why is my AGC block not adjusting the signal strength appropriately?
- Why is my ADSB-Decoder less sensitive than i would expect?

Telemetry could in these cases e.g. show, what the currently incoming power level of the IQ samples is, as well as the upper and lower threshold of the AGC, or its scaling factor. This gets even more interesting, when this data is visualized over time, because the behaviour of a block can be monitored, including how the AGC thresholds are adjusted over time or when e.g. edge cases are occuring.

# Implememntation Ideas
An option to collect the telemetry data could be [OpenTelemetry for Rust](https://github.com/open-telemetry/opentelemetry-rust) and the vendor agnostic [OpenTelemetry-Collector](https://github.com/open-telemetry/opentelemetry-collector).
Telemetry data can either be "pushed to" or "scraped by" a collector.
Afterwards, the telemetry data could be processed further with each persons favourite tooling in a pipeline like e.g. FutureSDR -> OpenTelemetry-Collector -> Prometheus -> Grafana.

The idea behind this issue is to discuss, if it would make sense to integrate a Telemetry Framework in the FutureSDR runtime.
Questions that have to be answered are:
- How can we give block implementors an easy interface within a block to export required telemetry data?
- Can we reuse the control port to automatically provide a /metrics route which can be scraped?
- How can we implement telemetry with the least performance degradation possible (e.g. default to no telemetry at all, using feature flags)?

Looking forward to some feedback! :)

bastibl, 2023-05-25:
Cool, idea . I was also missing something like that many times.

Just some first thoughts:
I think the main challenge here is to strike the balance between providing observability and keeping the complexity of the runtime at a manageable level. Maybe this can be piggybacked on top of PMTs: For example, a block can optionally define a state that is in essence a map String -> PMTs and the data in this map is accessible from other tasks that are spawned on the runtime. This could be very similar to the `BlockDescription`, which can already be polled through the `Flowgraphhandle`. So maybe add a `BlockState` to just get the state.

There could then, for example, run an OpenTelemetry task on the async runtime that sends this data on demand or periodic to an outside service, while doing the PMT -> whatever format conversion. This way, there could be alternate implementations and the FutureSDR runtime wouldn't have to be tight to this particular implementation. The same information could, for example, be used by a web GUI that allows to click on the blocks of a running flowgraph to get further information about it's state.

I wouldn't mind too much about portability and performance for now. This would have to be feature-gated so that there is no (or nearly no) overhead when it is not enabled. And on architectures like the web, this feature will likely not be available (at least not from the start).

ratzrattillo, 2024-05-22:
After letting this issue rest for one year, i will try to tackle it again.

As a recap the differences of Logging, tracing and metrics in form of examples:

**Log Example**:
- FutureSDR Starting up
- Opening Control Port
- Flowgraph Starting up

**Metrics Example**:
- Internals of AGC Block at given point in time like e.g.: {timestamp: xyz, input_signal_power: 0.3, output_signal_power: 1.0, scaling_factor: 3.33, step_size: 0.1234}

**Traces Example**:
- A batch of I/Q samples is monitored during its processing in a flowgraph, entering Block after Block (Flowgraph starts top span, while the processign of the batch in each block is a subspan)

Still, support for OpenTelemetry in Rust is mostly in Alpha stage: https://opentelemetry.io/docs/languages/rust/#status-and-releases
As the APIs change fast, I suppose it is best to have telemetry support in a seperate subcrate, where the interface we want to offer to the futuresdr users is abstracted and potentially more stable.

We also have to think about where we want to have telemetry. In the blocks or in the runtime, or both? This affects the architecture of FutureSDR aswell.
The goal should be to keep telemetry the least intrusive as possible, so not a lot of areas in the FutureSDR code are cluttered with it.

As telemtry collection is costly, it should be possible to define which metrics/traces/logs are collected at runtime. Especially in Blocks with high sample rates, this could lead to an immense amount of collected data.
I also believe, that connection details to OpenTelemetry Collectors should be definable in the Config.toml of FutureSDR.
It could also be possible to define logging/tracing/metrics when initializing the runtime like e.g.:
```rust
Runtime::new().with_otel_logger(logger_config).with_metrics([AGCBlock::Metric1, AGCBlock::Metric2, FIilterBlock::MetricX,...]).with_traces([Trace]);
```

It still is difficult for me to imagine a way, which will not clutter the blocks or runtime with telemetry code.

I made my first tries on this branch: https://github.com/ratzrattillo/FutureSDR/tree/telemetry

Looking forward to some feedback and discussion!

bastibl, 2024-05-27:
Hi, thanks for continuing this work. That's great. And having telemetry in a separate crate and feature-gated in the main project is the way to go.

I don't have any experience w/ this topic, so it will take me some time to get into it. I should have more time to work on FutureSDR and 2-3 weeks. Then I'll have a more in-depth look into it.

srsRAN also has something like this. Maybe it makes sense to see how they implemented it: https://docs.srsran.com/projects/project/en/latest/user_manuals/source/grafana_gui.html

ratzrattillo, 2024-06-30:
Up till now, it seems like Grafana is useful for logging FutureSDR log output.
I haven't discovered a very good use case for the traces fucntionality within the FututreSDR runtime yet.
The metrics helped me quite a lot in debugging e.g. the behaviour of the AGC block.
However, until now the efficiency in collecting the metrics and their resolution might not fit the needs of applications that make use of high sample rates. But for e.g. the CW examples, it already does really fine!

Created a merge request for my current work so far at #140.

metasim, 2024-11-14:
Since this crate already uses `tracing`, and the user can select [OpenTelemetry][1] as the subscriber, wouldn't additional instrumentation of calls be sufficient integration to support OTEL? Or is it more complicated than I'm giving credit for? 

IMHO, I'd keep the instrumentation goals separate from the collection infrastructure, with collection being application-specific and outside the scope of this library. For example, I've had really great experience with [Honeycomb][2] in the past, which is *not* OTEL, but has a `tracing`-compatible subscriber.

That said, properly instrumenting the FutureSDR code (ensuring performance isn't affected, and requisite information is included), is an initiative that could serve the community well, long term.

[1]: https://docs.rs/tracing-opentelemetry/latest/tracing_opentelemetry/
[2]: https://www.honeycomb.io

closed 2026-10-04 by ratzrattillo