FutureSDR/FutureSDR #116: Terminate `AudioSource` when the end of the file is reached

corvus-ch, 2023-07-11:
I try to write a flow graph reading an audio file, do some processing and writing the result to another file. Such a flow graph never exits, and the output is growing and growing. Testing with an `AudioSink`, I figured that, once the file ends, the output is a continuous hum.

I can mitigate this by using `Head`, but this requires knowing the length in advance.

Is this by intention? If so, what would be the correct way to approach this? If not, I suggest changing the current behavior and make `AudioSink` terminate once the end of the file is reached.

bastibl, 2023-07-12:
Hi, can you try the `main` branch from [this repo](https://github.com/bastibl/FutureSDR). To test it, you can change the FutureSDR dependency in your `Cargo.toml` like this: https://doc.rust-lang.org/cargo/reference/specifying-dependencies.html#specifying-dependencies-from-git-repositories

Generally speaking: 
- in debug mode, blocks should indicate when they terminate
- if a block is not terminating, it's probably because there is missing (or broken) logic to set `io.finished = false` in the blocks work function.

See this commit, for example: https://github.com/bastibl/FutureSDR/commit/08be115e032d3de29445068541393e42f7ff7bb5#diff-bbcb681b01f127c9e196e087099421568c7198032bdc8720ebe229d62200585eL58-L62

corvus-ch, 2023-07-12:
Using commit `95ebedc`  (latest commit on branch `main` as of the time of this writing) of the repository https://github.com/bastibl/FutureSDR works as expected.

Am I right, assuming that this will be merged here eventually and become available in an upcoming release?

bastibl, 2023-07-12:
Thanks for testing. I merged it into main and made a new release 0.0.35.

corvus-ch, 2023-07-12:
Many thanks.

closed 2023-07-12 by corvus-ch