FutureSDR/FutureSDR #117: Example for SSB modulation

corvus-ch, 2023-07-15:
Recently, I played around with FutureSDR. That resulted in me implementing SSB modulation. Would it be interesting for this project to have this added as an example? If that is the case, I happily would provide a PR.

bastibl, 2023-07-17:
Sounds great! There's also a [limited SSB](https://github.com/FutureSDR/FutureSDR/tree/main/examples/ssb-receiver) as an example. Maybe it could be incorporated?

But, to be honest, the best case would be, if this could live outside of the main repo. Since otherwise, I'm the single point of failure :-) I'd be happy to prominently feature external projects in the readme and the website.

loic-fejoz, 2023-07-18:
I would propose to push the main blocks to `fsdr-blocks` and have your main example in your on repository.
NB: May you propose a  block, the goal is to keep similar hiearchy as in GNURadio.

Rationale for having the example in your on repository is that it is hard to maintain a real portable one. So unless you are ready to maintain it, it clarifies the expected state of the project.

Note also that it is already possible to have an SSB decoder with `fsdr-cli`
NB: I will also move some blocks from `fsdr-cli` to `fsdr-blocks`.

corvus-ch, 2023-07-30:
I have done an initial implementation in #119. Please advice if this is going in the right direction or not.

bastibl, 2023-08-25:
Should be fixed with #119

closed 2023-08-25 by bastibl