FutureSDR/FutureSDR #120: choosing hackrf as device for examples?

RegEM, 2023-09-25:
Hi Everyone,
Do I need to choose or set hackrf as my device for the examples?
I get an error, 
FutureSDR: ERROR - Source_0: Error during initialization. Terminating.
FutureSDR: ERROR - Source_0: Error in Block.run() Soapy (StreamError: RtAudio init error 'RtApiDs::probeDeviceOpen: error (Invalid parameter) creating input buffer (Default Device)!)
Richard

RegEM, 2023-09-25:
I'll add that I'm on windows 10, and I can connect to my hackrf using CubicSDR.

RegEM, 2023-09-25:
I guess I do have the other error,
←[1m←[33m[WARNING] SoapyVOLKConverters: no VOLK config file found. Run volk_profile for best performance.←[0m

bastibl, 2023-09-25:
Soapy considers your sound card as SDR. Most applications have CLI parameters to select the soapy driver.
` cargo run --release --bin [...] -- -a soapy_driver=hackrf`

RegEM, 2023-09-25:
Thanks Bastian. That works fine. 
I am receiving 'Seify Source Overflow' with the keyfob and adsb examples, but confirmed I'm headed in the right direction by connecting error free to the fm-receiver example.  Always nice to hear the sdr radio...
Cheers, Richard

closed 2023-09-25 by RegEM