FutureSDR/FutureSDR #121: No constellation modulator?

xpromache, 2023-10-04:
As a newbie I found the gnuradio constellation modulator useful. 

Is that considered for FutureSDR?

bastibl, 2023-10-04:
AFAIK, there is none at the moment (maybe @loic-fejoz knows better?). A simple one could be implemented on-the-fly using, for example, [ApplyNM](https://docs.rs/futuresdr/latest/futuresdr/blocks/struct.ApplyNM.html) block followed by an RRC filter.
QPSK, would look something like this, followed by a pulse shaping filter:

```rust
    let modulator = ApplyNM::<_, _, _, 1, 4>::new(|byte: &[u8], symbols: &mut [Complex32]| {
        const A: f32 = std::f32::consts::FRAC_1_SQRT_2;
        for i in 0..4 {
            let sym = match (byte[0] >> 2 * i) & 0b11 {
                0 => Complex32::new(A, A),
                1 => Complex32::new(-A, A),
                2 => Complex32::new(A, -A),
                _ => Complex32::new(-A, -A),
            };
            symbols[i] = sym;
        }
    });
```

I'm happy to support if you want to give it a try, but I won't have the time to do it myself in the next weeks.

loic-fejoz, 2023-10-05:
At the moment, there are no such block in [fsdr-blocks](https://github.com/FutureSDR/fsdr-blocks) but it would perfectly make sense to have it, for sure! In fact, the goal is to have nearly equivalent blocks for every things.
I would indeed start with the code @bastibl propose and then multiply with a signal source.

@xpromache if you already have some existing GNURadio workflow that you would like to see executed with FutureSDR, I would be interested for you to share for my project [fsdr-cli](https://github.com/loic-fejoz/fsdr-cli).

loic-fejoz, 2023-10-05:
By the way, I was thinking of multiplying by a signal source in case you wanted more than one sample per symbol.
But it is overkill... I just had a look at the [GNURadio Constellation code](https://github.com/gnuradio/gnuradio/blob/main/gr-digital/lib/constellation.cc#L126) and they are simply using a pre-computed table.
Also have a look at [generic_mod_demod.py](https://github.com/gnuradio/gnuradio/blob/main/gr-digital/python/digital/generic_mod_demod.py) to see the exact blocks the constellation block is using hierarchically.

xpromache, 2023-10-05:
Hey, thanks for your answers. I don't have any workflow yet, I just started with signal processing and trying to wrap my head around everything I read in the last month. I will try to do the constellation modulator block but it will take me a while.

A bit off topic since the two of you are in the "trade" for a long time. I'm looking to make a OQPSK demodulator and I read different opinions on how the blocks should be structured with the symbol sync before or after costas loop and with a matched filter somewhere in between. But now I have seen in the book "Digital Communications a Discrete Time Approach" a scheme which frankly matches most with the theory I read so far:
[image]

The only problem is that this mixes up the symbol sync with the carrier sync with the matched filter and polyphase resampler in a big loop, so it seems like impossible to implement in GNURadio or FutureSDR.
What is your take on this?

bastibl, 2023-10-06:
Such feedback-based systems are difficult/not efficient to do in a stream processing system. GNU Radio, therefore, does this in larger/complex/monolithic blocks. See this, for example: https://github.com/gnuradio/gnuradio/blob/main/gr-digital/lib/symbol_sync_ff_impl.cc

I guess that's the way to go.

bastibl, 2023-10-10:
For most cases (i.e., parameters) the rational resampler should be fine. The difference is that the rational resampler supports only rates `a/b` where `a` and `b` are integers. The arbitrary rate resampler can resample by any rate, such as `pi`. Implementing the arbitrary rate resampler would be great, but it's rather complicated, since it also requires polyphase filter banks. I would recommend it as the first project.

closed 2024-09-10 by bastibl