FutureSDR/FutureSDR #125: The FIR filtering seems to not work with Complex32 signals

MarSik, 2023-12-29:
I have tried to implement a 2-FSK transmitter using the FutureSDK pieces (+ a VCO of my own), but whatever I try I cannot get a clean signal output.

[image]

Please check out the code here https://github.com/MarSik/debug-futuresdr-fsk-filtering/tree/no-filtering-issue

The debug outputs all show proper modulation at +/- 45 kHz, but the harmonics are still there. Comparing the output pre-filtering and post-filtering shows no difference.

The filter (140 kHz low pass with 50 kHz transition bw) I use:

```
let roofing_filter_taps = futuredsp::firdes::kaiser::lowpass::<f32>(140e3_f64 / SAMPLE_RATE, 50e3_f64 / SAMPLE_RATE, 0.01);
    let roofing_filter = FirBuilder::new::<Complex32, Complex32, f32, Vec<f32>>(roofing_filter_taps);
```

Sampling rate is 8 MHz.

I might be doing something wrong, but I can't figure out what that is.

The same approach (VCO + low pass filter) works just fine in GNURadio.

[image]

AndersKaloer, 2023-12-30:
Hi, I spent some time looking into this problem.
In the end, I realized that Inspectrum treats the ".c32" extension as signed integer samples.
You get comparable results if you use ".cf32" :-)

See more here: https://github.com/miek/inspectrum

Note, however, that there can be some differences in the outputs produced by FutureSDR and GNU Radio, since the filter taps are generated using different techniques.

MarSik, 2023-12-30:
Oh, .. thank you very much for figuring that out. I totally missed that part of the inspectrum README. I did not want to transmit noise to the air.

Yeah, the small filter differences are expected and not an issue.

Thanks a lot once more, all seems to work as it should indeed.

closed 2023-12-30 by MarSik