FutureSDR/FutureSDR #128: Accessing data in VectorSink

klangner, 2024-01-27:
I'm building simple GUI app for showing the spectrum. Just for learning purposes.
As a starting point I have Source -> FTT -> VectorSink

Now I need to access samples from VectorSink while the flowgraph is working. VectorSink kernel has function `items()`, but since I'm calling it from different thread I can't use this function directly.  I think I should use message to get this data.

Currently VectorSink doesn't support this as an output. Would it make sens to add it to it? Something like
```
StreamIoBuilder::new()
  .add_input::<T>("in")
  .add_input("items", Self::items_handler)
  .build()
```
Where the message would be `Pmt::VecF32`. 
The handler would copy internal buffer into a given vector.

Or maybe there is another way to get the output from FFT in realt time which I'm missing?

JuliDi, 2024-01-27:
There is the web socket example https://github.com/FutureSDR/FutureSDR/blob/main/examples%2Fwebsocket.rs with a sink where you could just dump the data. I've built a small "real-time" plotting tool with plotters and iced a while ago that used the web socket sink on the SDR side to stream the FFT Results. Then the GUI would update the plot on every new FFT result buffer.

Never managed to make it nice and publish it, but I can share the code if you're interested in that.

klangner, 2024-01-27:
If you could share this example that would be great, thank you!

Still wouldn't it be useful to have function to pull data instead of using tcp to push data between local processes?

JuliDi, 2024-01-27:
Added you to the repo, you _should_ (haven't used it in a while) be able to just run it and use futuresdr to stream data into a web socket. Hope it helps.

Do you mean independent processes or different threads in the same program? If you want to get data out of your application, I guess some sort of network socket (possibly UDP even) is the most platform independent and easiest to use way to do it. And the overhead is manageable. Especially on Linux, there are some other ways afaik like writing into something like file streams.

If you have a specific proposal for how to implement this, I'd love to hear about it. Looking into doing something similar in another project at the moment.

Also, if it's just another block for futuresdr I don't think there is a reason why we shouldn't add it.

klangner, 2024-01-27:
I mean different threads in the same app. Here is my example (which doesn't compile)
https://github.com/klangner/yasa/blob/1f27b37c30564d9705e1f12649154fe46b01ea2a/spectrum/src/main.rs

I'm using egui which refreshes UI 60 times per second. And I would like to be able to pull data fast to not slow down redraw. (I can read data every 10th frame if needed).

I will give it a try with web sockets. Thanks for the code!

But if it would be ok, to add input `items` to the VectorSource to fetch data, I'm willing to do it as a PR.
Something which will allow to use api like:
```rust
async_io::block_on(handle.call(
                            id,
                            items_port_id,
                            Pmt::VecF32(&buffer),
                        ))?;

klangner, 2024-01-28:
Yes you do understood it correctly. And I know I can easly solve it with my own Sink.

My issue was question about adding functionality to the VectorSink. And the example which I gave was to present use case for it. I thought that VectorSink could be more useful if we could access its data from other threads. Since now the function `items` works only when the graph stops. 

But of course if this feature doesn't make sense, that's ok :-) We can then close this issue.

And @JuliDi, Thank you for your time and help!

JuliDi, 2024-01-28:
You're totally right, if the items function only works when the flow graph has terminated, it's of limited use. So I guess it would be good to have an extension (or new block) that can be used to access the data at runtime.

loic-fejoz, 2024-01-28:
Clearly the VectorSink is not meant for that purpose.

Note that there is also the crossbeam sink/source that may help you.
https://github.com/FutureSDR/fsdr-blocks/tree/main/src/channel

Also, as you said, it is so easy to build your own sink dedicated to your own needs, that it is hard to have the exact sink you need.
That is why there is also the Sink that call your own callback for whatever purpose:
https://github.com/FutureSDR/FutureSDR/blob/main/src/blocks/sink.rs

klangner, 2024-01-28:
I see, thanks for the tips!.
In this case I think I will close this issue, since it is solved for me.

closed 2024-01-28 by klangner

bastibl, 2024-02-04:
I just added two small egui examples: One where the GUI is decoupled from the flowgraph (and could run on another host) and one where they are in the same process and can share memory.

See here for a quick video: https://x.com/bastibl/status/1754078443912966567?s=20