FutureSDR/FutureSDR #133: wasm example fails to build

JuliDi, 2024-03-23:
Using the current main branch, the wasm example fails to build. This also happens when I try to compile a wasm project with FutureSDR as a dependency. I will try to figure out what's the issue and report back. Any hints on where to look are welcome!

```
error[E0433]: failed to resolve: could not find `UsbControlTransferParameters` in `web_sys`
   --> /home/user/Git/FutureSDR_main/src/blocks/wasm/hack_rf.rs:249:34
    |
249 |         let parameter = web_sys::UsbControlTransferParameters::new(
    |                                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^ could not find `UsbControlTransferParameters` in `web_sys`
```

This indicates that web-sys is somehow not working, maybe a missing feature.

JuliDi, 2024-03-23:
Right after opening the issue I may have found it. The web-sys docs say that `RUSTFLAGS=--cfg=web_sys_unstable_apis`  is required, see https://rustwasm.github.io/wasm-bindgen/api/web_sys/enum.UsbRecipient.html

I will make a PR that adds a .cargo/config.toml file that sets these rustflags.

closed 2024-03-24 by bastibl

bastibl, 2024-03-24:
Unfortunately, the `wasm` example still used the older build system. I just ported it to trunk, which is much nicer and requires less javascript (i.e., no node dependency and no custom javascript on the website). You can try it with `trunk server` or `trunk serve --release`.

PR #135 (2024-03-24): Fix wasm example
Close #133 by adding compiler flags