FutureSDR/FutureSDR #137: Error connecting input to PulseAudio server, Ubuntu24 .

springyworks, 2024-05-17:
ERROR below is produced on Ubuntu24. This version 24 of ubuntu (linux) uses pipewire on top of the ALSA sound system. The more older PulseAudio sound service is replaced by an emulator (by default) , see [pipewire-pulse](https://docs.pipewire.org/page_man_pipewire-pulse_1.html)  command/service . Maybe Seify needs a wash ;-)

```
~/rust/repos/FutureSDR/examples/egui$ cargo run --bin combined
    Finished `dev` profile [unoptimized + debuginfo] target(s) in 0.22s
     Running `target/debug/combined`
[INFO] [UHD] linux; GNU C++ version 13.2.0; Boost_108300; UHD_4.6.0.0+ds1-5.1build4
[INFO] Using format CF32.
[2024-05-17T13:08:57Z ERROR futuresdr::runtime::block] Source_0: Error during initialization. Terminating.
[2024-05-17T13:08:57Z ERROR futuresdr::runtime::block] Source_0: Error in Block.run() Soapy (StreamError: RtAudio init error 'RtApiPulse::probeDeviceOpen: error connecting input to PulseAudio server.)
```

N.b. I tried to debug this with LLDB and rust-analyzer, but it goes over my head at the moment.  I got the `audio` and `rattlegram` executables working ! 

Anyone interested in HAM radio weaksignal digital modes , let me know!  I operate with HF 10W in a tuned slanted asymmetric dipole.
PE1BFE 
Ko
Afrt NL

K4HVH, 2024-05-24:
I'm having the same issue, +1

springyworks, 2024-05-29:
with the right frequency and sample_rate , the egui `combined` example gets a life! I see an updating spectrum, but a bit choppy. 
Also see this [seify issue](https://github.com/FutureSDR/seify/issues/8#issue-2303657703)
```
.....
            let src = SourceBuilder::new()
                .frequency(0.0)
                .sample_rate(48000.0)
                .gain(34.0)
                .build()?;
......
```
RtAudio pulse: _NOT_ running realtime scheduling, probably comes from a mismatch in buffersize ?

bastibl, 2024-05-29:
I assume because you have to give your user privileges to configure a process to set real-time priority.
I have in `/etc/security/limits.conf.d/limits.conf`
```
basti            -       rtprio          99
```
Depending on your distribution it might also put some config in `/etc/security/limits.d/audio.conf` to set real-time priority for every member of the group `audio`. In that case, you could just put your user in the audio group.

But that's really an issue of RtAudio. Maybe they have some documentation on this?

springyworks, 2024-05-29:
that helped ! running realtime now!

the graph visual updates still stay a bit irregular , but i can live with that.

bastibl, 2024-05-29:
Great, I'll close this issue for now. Feel free to reopen, if there are still problems.

closed 2024-09-10 by bastibl