FutureSDR/FutureSDR #138: How to pick between two different Soapy Devices

bwhetsel, 2024-05-23:
SourceBuilder::new() is picking the wrong device, how can I ensure the device I want to pick?

JuliDi, 2024-05-24:
If I understand correctly what you're doing (some more context and maybe a few lines of code would be helpful here), you should be able to set the device with this function on the builder: https://github.com/FutureSDR/FutureSDR/blob/66b11d599595e6a11f3bf1bcc0935eb50f5f3ff8/src/blocks/seify/builder.rs#L45

bwhetsel, 2024-05-24:
Thank you that helped narrow my problem. The primary difficulty I am facing is that the seify “probe” example works and shows the device I want to use. When I use futureSDR re-export of seify::enumerate() with the same hardware setup I get different behavior (soapy hackrf device is not listed).

JuliDi, 2024-05-25:
Interesting. Are you using the same version, i.e., are you using the main branch for everything?

bastibl, 2024-05-26:
You could also do `SoapySDRUtil --probe` to see what Soapy find. Often the OS installs all available Soapy modules which can be a pain. For example, since it always regards any audio card as "SDR" and it might get picked. To avoid this, you could uninstall the annoying Soapy modules.

You can, however, also specify the driver as argument to Seify (see the [WLAN example](https://github.com/FutureSDR/FutureSDR/blob/main/examples/wlan/src/bin/rx.rs#L33-L35)), i.e., `soapy_driver=uhd`. Or as CLI parameter `-a soapy_driver=uhd`. It is also shown here: https://youtu.be/NytXsgbv-PE?si=23t8KrM02wYVrf6z&t=1390

closed 2024-09-10 by bastibl