FutureSDR/FutureSDR #169: User-provided value to `BlockMeta::instance_name` is not preserved after insertion into `Flowgraph`.

metasim, 2024-11-12:
I was looking to use the `BlockMeta::instance_name` as a means of assigning application-specific unique identifiers to particular stages in the flow graph for deterministic lookup via `Remote`. The following test fails, indicating the user-specified instance name is overwritten by the internals, making the `Block::set_instance_name` method unuseful.

## Test

```rust
use futuresdr::anyhow::Result;
use futuresdr::blocks::NullSink;
use futuresdr::blocks::NullSource;
use futuresdr::runtime::Flowgraph;
use futuresdr::runtime::Runtime;
use futuresdr_macros::connect;

#[test]
fn flowgraph_instance_name() -> Result<()> {
    let rt = Runtime::new();
    let name = "my_special_name";
    let mut fg = Flowgraph::new();

    let mut source = NullSource::<f32>::new();
    let sink = NullSink::<f32>::new();
    source.set_instance_name(name);
    connect!(fg, source > sink);
    let (th, mut fg) = rt.start_sync(fg);

    let desc = rt.block_on(async move { fg.description().await })?;
    assert_eq!(desc.blocks.first().unwrap().instance_name, name);
    Ok(())
}
```

## Culprit

Line 124 below overwrites the user-assigned value.

https://github.com/FutureSDR/FutureSDR/blob/34bf04f7447980f0b56ad8d407499dd4397c7010/src/runtime/topology.rs#L120-L126

## Possible solution

* Only set instance_name in `Topology::add_block` when it is provided as `None`.
* If `Block::instance_name()` is `Some(..)`, then check `Topology::blocks` for instance name uniqueness.
  - `Topology::add_block` does not return `Result` so the only error handling option is to `panic`, or break the API, or issue a log warning.
  - Implication of the uniqueness constraint not being held are `Remote` queries for a `Block` by name make provide unexpected results.

metasim, 2024-11-12:
I'm happy to implement a fix if given guidance on the approach that should be taken here.

bastibl, 2024-11-13:
Code created by the connect macro already uses `?`, so this shouldn't be an issue, no?

https://github.com/FutureSDR/FutureSDR/blob/bf3aaaf3ba58e102eb9a43c94e664d09c8520fdb/crates/macros/src/lib.rs#L203

metasim, 2024-11-13:
I'll take a stab at a fix.