FutureSDR/FutureSDR #175: Remove re-export of `futuresdr::anyhow`

metasim, 2024-11-14:
[Per convention][1], it is preferred to have library specific error types (usually via `thiserror`), and to use `anyhow` for generic application/binary error handling. 

In the context of `futuresdr`, this means removing the use of `anyhow` in the public API, as well as the `futuresdr::anyhow` re-export. The examples will continue to use `anyhow`. Non-public APIs may continue to use `anyhow` when expedient.

See [Discord conversation][2] for additional context.

[1]: https://www.shakacode.com/blog/thiserror-anyhow-or-how-i-handle-errors-in-rust-apps/#:~:text=thiserror%20is%20a%20derive%20macro,default%20error%20type%20in%20applications.
[2]: https://discord.com/channels/856229400085528637/856229400525144065/1306711785851850782

bastibl, 2024-11-15:
I started to work on this but noticed a few things that changed my mind regarding the re-export. The rest is true, and we already converted some (or all) public APIs to proper errors, I think `anyhow` should only be used by the `Kernel` functions (init, work, deinit), where it makes sense to ease block development. However, this use in the core library also speaks for a re-export of the crate:

- It's likely that apps that use FutureSDR will use `anyhow` to ease error handling. Basically all of the examples do. While the apps  could just add it as a dependency, `anyhow` is also used internally by FutureSDR and these versions might diverge, causing problems. I had a similar issue with `num_complex` some time ago.

- `anyhow` is used by core FutureSDR APIs (mainly the `Kernel` functions) that the user wants to implement for custom blocks. Without a re-export, the user has to manually check the FutureSDR code for the anyhow dependency and make sure they use a compatible version in their own module.

- `anyhow` is referenced in macros (e.g. https://github.com/FutureSDR/FutureSDR/blob/main/crates/macros/src/lib.rs#L644), which make the dependency non-obvious in the code, i.e., a user might not understand why they have to depend on `anyhow` even though there is no obvious use of the crate in their code.

So I think we should keep the re-export but continue to convert to proper errors.

metasim, 2024-11-15:
That makes sense. If you're writing a `Block` and use `anyhow::Error` in your application code, you'd want to use the same version of `anyhow` as `futuresdr`.

Another option is to type alias `anyhow::Error` as something crate specific (I don't know, `futuresdr::MiscError`), but that starts to feel like bike-shedding.

bastibl, 2024-11-15:
But just calling it differently with a type alias wouldn't solve the version issue, no?

metasim, 2024-11-15:
Correct, but it would make it easier to change the exposure level to `anyhow` in the future. Probably not worth worrying about at this stage of development. I'm just in a habit of always having an `error.rs` file in all my crates where I either type alias `Result` and `Error` to something else (usually `anyhow`), or define my own (via `this error`).

PR #177 (2024-11-19): Introduce a custom Result Type
This hides the `anyhow::Result` in the public API (mainly message handlers and Kernel functions) through a type alias and removes the `anyhow` re-export.

fixes #175