FutureSDR/FutureSDR #181: Cryptic panic in terminate_and_wait if Runtime dropped early

TroyNeubauer, 2024-11-24:
I was experimenting with flowgraph shutdown today and hit a strange panic in the standard library caused by `terminate_and_wait`. Looks like this only happens if you call `fg.terminate_and_wait()` after the `Runtime` is dropped:

```
Press enter to stop

2024-11-24T16:30:29.193070Z DEBUG smol-29 futuresdr::runtime::block: FileSink-2 terminating
2024-11-24T16:30:29.193070Z DEBUG blocking-1 futuresdr::runtime::block: Source-0 terminating
2024-11-24T16:30:29.193196Z DEBUG    smol-31 futuresdr::runtime::block: Fir-1 terminating
thread 'smol-29' panicked at std/src/sys/pal/unix/thread.rs:302:9:
failed to join thread: Resource deadlock avoided (os error 35)
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
thread 'smol-31' panicked at /home/troy/foxhunter/FutureSDR/src/runtime/scheduler/smol.rs:84:61:
called `Result::unwrap()` on an `Err` value: Canceled
thread 'smol-30' panicked at smol worker panicked /home/troy/foxhunter/FutureSDR/src/runtime/scheduler/smol.rsErr:(84Any: { .. }61):

called `Result::unwrap()` on an `Err` value: Canceled
smol worker panicked Err(Any { .. })
```

Minimal code to reproduce:
```
use futuresdr::{
    blocks::{seify::SourceBuilder, FileSink, FirBuilder},
    macros::connect,
    num_complex::Complex32,
    runtime::{Flowgraph, FlowgraphHandle, Runtime},
    seify::{Device, DeviceTrait},
};

pub fn start<D: DeviceTrait + Clone>(
    device: Device<D>,
) -> Result<FlowgraphHandle, Box<dyn std::error::Error>> {
    let frequency_hz = 92_300_000.0;
    let sample_rate_hz = 2_500_000.0;

    let mut fg = Flowgraph::new();

    let src = SourceBuilder::new()
        .device(device)
        .frequency(frequency_hz)
        .sample_rate(sample_rate_hz)
        .build()
        .expect("failed to create seify device");

    let resamp = FirBuilder::resampling::<Complex32, Complex32>(20_000, sample_rate_hz as usize);
    let snk = FileSink::<Complex32>::new("out.cf32");

    connect!(fg, src > resamp > snk.in;);

    // Start the flowgraph and save the handle
    let rt = Runtime::new();
    let (_res, flowgraph) = rt.start_sync(fg);

    Ok(flowgraph)
    // rt is dropped here, and we are returning a handle to the flowgraph running inside of it
}

fn main() {
    let dev = futuresdr::seify::Device::new().expect("Failed to open radio");

    // Call function which only returns flowgraph handle and not runtime
    let mut handle = start(dev).unwrap();

    println!("Press enter to stop");
    let mut line = String::new();
    std::io::stdin()
        .read_line(&mut line)
        .expect("Failed to read from stdin");

    // Somewhere else, in the codebase, use the stored flowgraph handle to terminate it
    futuresdr::async_io::block_on(handle.terminate_and_wait()) // Causes panics
        .expect("Failed to shutdown flowgraph");
}
```

If I create the runtime beforehand, and make sure it isnt dropped while the flowgraph is running, it shutdowns successfully.

```
use futuresdr::{
    blocks::{seify::SourceBuilder, FileSink, FirBuilder},
    macros::connect,
    num_complex::Complex32,
    runtime::{scheduler::SmolScheduler, Flowgraph, FlowgraphHandle, Runtime},
    seify::{Device, DeviceTrait},
};

pub fn start<D: DeviceTrait + Clone>(
    rt: &Runtime<SmolScheduler>,
    device: Device<D>,
) -> Result<FlowgraphHandle, Box<dyn std::error::Error>> {
    let frequency_hz = 92_300_000.0;
    let sample_rate_hz = 2_500_000.0;

    let mut fg = Flowgraph::new();

    let src = SourceBuilder::new()
        .device(device)
        .frequency(frequency_hz)
        .sample_rate(sample_rate_hz)
        .build()
        .expect("failed to create seify device");

    let resamp = FirBuilder::resampling::<Complex32, Complex32>(20_000, sample_rate_hz as usize);
    let snk = FileSink::<Complex32>::new("out.cf32");

    connect!(fg, src > resamp > snk.in;);

    let (_res, flowgraph) = rt.start_sync(fg);

    Ok(flowgraph)
}

fn main() {
    let dev = futuresdr::seify::Device::new().expect("Failed to open radio");
    let rt = Runtime::new();

    let mut handle = start(&rt, dev).unwrap();

    println!("Press enter to stop");
    let mut line = String::new();
    std::io::stdin()
        .read_line(&mut line)
        .expect("Failed to read from stdin");

    // Somewhere else, in the codebase, use the stored flowgraph handle to terminate it
    futuresdr::async_io::block_on(handle.terminate_and_wait()) // Now fine
        .expect("Failed to shutdown flowgraph");

    // rt dropped here, after flowgraph has finished
}
```

In hindsight, its obvious that dropping the runtime supporting an active flowgraph is a bad idea, but I think this is a bit of a footgun, since it looks like most examples create the runtime inline with the other variables. Its also unintuitive since you only get the panic when cleaning up, not when the Runtime is dropped.  

If one naively pastes an example into their own function that returns a flowgraph handle (and implicitly drops the Runtime), they will hit this.

The error seems to be caused by a thread trying to join itself. I poked around a bit in the internals, but couldnt figure out how to patch this to handle gracefully. For now making sure the Runtime outlives any flowgraph it runs works well enough.

bastibl, 2024-11-25:
Thanks for the detailed issue. I fully agree that this is a footgun, especially since there is no indication about the underlying problem.

Some more context: The `start` function of the `Runtime` returns a `TaskHandle` and a `FlowgraphHandle`. We already had similar issues with the task handle in the past. Here, we solved it by introducing (artificial) lifetime bounds that would result in a compile error, if the runtime is dropped too early.
I first though about introducing a similar mechanism for the flowgraph handle, but I think that does not make sense. The flowgraph handle is typically cloned and passed around to other tasks/threads to be used as an RPC-like interface to the flowgraph. This does not work with lifetime bounds. So I would argue that it is OK that the broken code compiles.
But obviously there shouldn't be this cryptic runtime error. The underlying issue is the following: while the `Runtime` is dropped after the function returns, the scheduler (i.e., task executor) still exists, since there is a reference to it in the task that is running the flowgraph (and I think that's fine too). Now when the handle is used to terminate the flowgraph, the last reference is dropped. But this happens inside an async task that is executed inside a worker thread of the executor that is about to get shut down. In the `drop` function, the scheduler wants to join worker threads, which means that one of the worker threads called join on itself, which leads to a panic: https://doc.rust-lang.org/std/thread/struct.JoinHandle.html

I added a check for this in the drop implementation. And also added a debug message when the runtime is dropped. I think this is the best we can do in this case. https://github.com/FutureSDR/FutureSDR/commit/4f63c2c4fe58ecf1739e7b4e70c9bb819e7c32aa

TroyNeubauer, 2024-11-25:
Makes sense. Thanks!

closed 2024-11-25 by bastibl