FutureSDR/FutureSDR #186: Ability to "take" value out of `PmtAny`

metasim, 2024-12-12:
It would be a nice convenience to be able to extract the value inside a `Pmt::Any` to enable a `Block` to take full ownership of the message contents without a post-downcast `clone`.

## Use Case

```rust
let p: Pmt = ...;
let my_foo: Box<Foo> = match p {
    Pmt::Any(payload) => payload.downcast::<Foo>(),
    _ => panic!(),
};
```

## First Attempt

We currently have this:

https://github.com/FutureSDR/FutureSDR/blob/9d264b27e01970a13ba4f9e24cdc5259227999e0/crates/types/src/pmt.rs#L36-L45

We could extend it with a `downcast` method _if_ there were a way to get a `Box<dyn Any>` (not `Box<dyn PmtAny>` via the `PmtAny` trait.

```rust
impl dyn PmtAny {
    ...
    pub fn downcast<T: PmtAny>(self) -> Option<Box<T>> {
        // if `fn to_any_boxed(self) -> Box<dyn Any>` existed on `PmtAny`
        self.to_any_boxed().downcast().ok()
    }
}
```

I assume this is possible, but I couldn't figure it out (without digging into `unsafe` methods).

metasim, 2024-12-12:
PS: I may be missing a more straightforward way of doing this, but right now I'm having to `clone` my received `Pmt::Any` to transfer ownership.

bastibl, 2024-12-13:
Can you try:
```rust
pub fn take<T: Any>(self: Box<Self>) -> Option<Box<T>> {
    let i: Box<dyn Any> = Box::new(self);
    i.downcast().ok()
}
```
I guess there is no way around the other boxed pointer.

Looks like it will be possible at some point but not today (see https://github.com/rust-lang/rust/issues/65991):
```rust
pub fn take<T: Any>(self: Box<Self>) -> Option<Box<T>> {
    let i: Box<dyn Any> = self;
    i.downcast().ok()
}
```

metasim, 2024-12-13:
Thanks for the tip! I had to do a bit a tiny bit more indirection to get it to work, but the suggestion above unblocked me. 🙏

PR #187 (2024-12-13): Adds `take<T: PmtAny>(self: Box<Self>) -> Option<Box<T>>`
Closes #186
commit 574827d0 (2024-12-13): Adds `take<T: PmtAny>(self: Box<Self>) -> Option<Box<T>>`
Closes #186
closed 2024-12-13 by bastibl