FutureSDR/FutureSDR #189: API control

randsoy, 2024-12-17:
I would like to modify more parameters and processes through the API (without disconnecting the WebSocket in between), for example:

Currently, the process is for ADS-B, but I want to change it to FM-receiver during use.
Also, the center frequency remains fixed, but the offset frequency can be adjusted.
I'm not sure if this is planned for the future or if this functionality is already implemented?

bastibl, 2024-12-19:
I'm not sure, if I understand the question. You could create a flowgraph with a branch for FM demodulation and a branch for ADSB and use a selector block to switch between them during runtime (https://github.com/FutureSDR/FutureSDR/blob/main/src/blocks/selector.rs).
And the SDR source has a handler to set the frequency: https://github.com/FutureSDR/FutureSDR/blob/main/src/blocks/seify/source.rs#L70
Is there anything else you would need for your application?

randsoy, 2024-12-19:
Thank you!Actually, I'm not very familiar with the futuresdr code, I only understand a little bit of Rust development, so I'll first look at the selector you mentioned and conduct some experiments.

randsoy, 2024-12-19:
The design of the selector is so beautiful. If my FM needs to adjust the offset frequency, how can I achieve it?

bastibl, 2024-12-19:
If you want to do offset tuning, you'd probably use the Frequency Xlating FIR: https://github.com/FutureSDR/FutureSDR/blob/main/src/blocks/xlating_fir.rs
However, at the moment, it lacks a message handler to set the offset frequency during runtime.

randsoy, 2024-12-23:
Do you have an example of 'Frequency Xlating FIR'? I tried to modify the offset and didn't get the translation offset I wanted. 'Apply::new' has the offset I want, but it can't be adjusted at runtime. Thank you!!!

bastibl, 2025-01-02:
The xlating FIR also doesn't have a message handler to adjust the offset during runtime, at the moment. Here is an example use in the code base: https://github.com/FutureSDR/FutureSDR/blob/main/examples/lora/src/bin/rx_meshtastic.rs#L85
If you have issues, it might be easiest to post your code.

closed 2025-04-25 by bastibl