FutureSDR/FutureSDR #191: Provide a canonical means of documenting ports, preferably in a runtime-queryable way.

metasim, 2025-01-08:
I would find it helpful if there was a canonical way to document ports: their purpose, and what messages or types they send/receive.

There are multiple levels of possible sophistication:

1. A documentation convention used throughout the library and serves as an example of how custom blocks should be described.
2. Extensions to `StreamIoBuilder` and `MessageIoBuilder` for providing simple text descriptions of inputs and outputs.
3. (2) with the addition of it included in the `Flowgraph` descriptor.
4. A macro that consumes `#[doc]` attributers and publishes them as a part of the `Flowgraph` descriptor.

Not sure if I'm the only one who struggles with keeping track of all the ports and their semantics, but I find discoverability a powerful aspect of FutureSDR, and this capability would greatly enhance that aspect.

bastibl, 2025-01-13:
Great idea. I was anyway thinking about changing some things in the runtime and that might align well. Creating a block will most likely require a proc macro annotation that also references the message handlers, i.e.,

```rust
#[block(message_handlers=...)]
struct MyBlock {
  ...
}
```

I didn't look much into the syntax that is typically used in those macros, but we could allow to (optionally) add a documentation string to the port. This info will then also become part of the remotely accessible [`BlockDescription`](https://docs.rs/futuresdr/latest/futuresdr/runtime/struct.BlockDescription.html) (which is part of the `FlowgraphDescrition`).

I'm not yet sure how to specify the handler. My current idea was something like
```rust
#[block(message_handlers=(foo, bar))]
struct MyBlock {
  ...
}
```

Which would use `Self::foo` and `Self::bar` as handlers with IDs according to the order in the tuple and the function name as port name.

If you have ideas how to incorporate a documentation string in such a macro that would be great. Maybe we could just support optional tuples as entries with `(function name, doc string)`.

metasim, 2025-01-13:
Oooh... great ideas. I think I see where you're going with this...

One of the things I like about `clap` and `utoipa` is that they treat `#[doc]` attributes as first-class entities (your doc comments become the runtime documentation). So that's the model I'm coming from.

I'll give it more thought, but one idea as I think out loud is to start with `message_handler` doing the heavy lifting, but would require changing the signature of `add_input`.

This:
```rust
    /// Returns whatever was sent.
    #[message_handler]
    async fn echo_handler(
        &mut self,
        _w: &mut WorkIo,
        _m: &mut MessageIo<Self>,
        _b: &mut BlockMeta,
        p: Pmt,
    ) -> Result<Pmt> {
        Ok(p)
    }
```

expands to:

```rust
    #[doc = "Returns whatever was sent."]
    fn echo_handler<'a>(
        _w: &'a mut WorkIo,
        _m: &'a mut MessageIo<Self>,
        _b: &'a mut BlockMeta,
        p: Pmt,
    ) -> (
        Pin<Box<dyn Future<Output = Result<Pmt>> + Send + 'a>>,
        HandlerMeta,
    ) {
        use futuresdr::futures::FutureExt;
        (
            Box::pin(async move { Ok(p) }),
            HandlerMeta::new("Returns whatever was sent"),
        )
    }
```

<details>
<summary>Assumes <code>HandlerMeta</code>...</summary>

<pre>
pub struct HandlerMeta {
    pub description: String,
    //... and other things... like expected input/output.
}

impl HandlerMeta {
    pub fn new(description: &str) -> Self {
        Self { description: description.into() }
    }
}
</pre>
</details>

Other options:

* Instead of the block constructor instantiating a `TypedBlock` with associated parts, use traits to describe the block, enabling the use of `derive` macros with additional field attributes for describing common scenarios. Then you'd have something like:
    - `TypedBlock:from<B>(bloc: B) -> ... where B: BlockMeta + StreamIo + MessageIo + Kernel`  
IOW, make it. closer to how `Kernel` works.
* The less "magical" approach would be to just extend `MessageIoBuilder` to have facilities for describing the handlers.

I'll keep thinking about it.

bastibl, 2025-04-25:
My current WIP code for FutureSDR creates blocks with a derive macro and supports message inputs like this:

```rust
#[derive(Block)]
#[message_inputs(my_handler)]
struct Foo {
  ...
}

impl Foo {
  pub async fn my_handler(...) {
  }
}
```

This new version actually does not need a `#[message_handler]` macro, since it does no longer have to box the message handler. The derive macro expands to a handler function that matches on the port ID and puts the PMT in the corresponding handler function. This allows us to get rid of [these crimes](https://github.com/FutureSDR/FutureSDR/blob/main/src/runtime/message_io.rs#L24-L34) and does not require boxing the handler future.

Now with regards to runtime documentation, there are two options:

1. Add it to the derive macro:

```rust
#[derive(Block)]
#[message_inputs(
    foo,                           // no doc
    bar = { doc = "Handles a bar event" },
    baz = { /* empty body = no doc */ },
)]
struct Foo {
  ...
}
```

This allows us to have separate doc strings that can be queried during runtime, e.g., `Foo::handler_doc("bar") -> String`. But these are independent from the doc string of the handler function.

2. Use handler macros to access doc strings
```rust
#[derive(Block)]
#[message_inputs(foo, bar, baz)]
struct Foo {
  ...
}

impl Foo {
  /// My Foo handler
  #[message_handler]
  pub async fn foo(...) -> {
}
}
```
impl Foo {

This way, we can access and store the doc string of the function and query it during runtime with a similar API, e.g., `Foo::handler_doc("foo") -> String`. It just means, we have to annotate handler functions just to access the docs. And API docs are the same as runtime docs.

What would you prefer? With both options, we can also extend the Block Descriptions so that remote interfaces can provide this information.

metasim, 2025-04-25:
Generally speaking, I think I'd first be interested in how all this would work _without_ macros, helping to inform the level of required complexity in the macro call site.

That is, are you leaning towards additional builder capabilities? Like:

```rust
MessageIoBuilder::new()
  .add_input("foo", Self::handle_foo)
  .with_description("Set foo with Pmt::f32; Request current setting with Pmt::Ok")
```

or is it more involved than that?

Knowing what the macro expands into might help determine what the best macro syntax is and where future enhancements might be needed.

---

That point aside, I think I like the second case better as it seems like there's less "macro magic" involved. I've been using Utoipa a lot lot lately, and it requires the use of massive attribute macros where syntax errors are easy and a bit tedious to deal with. So I might be overcorrecting, but I tend to think it's better to describe as much as possible using standard Rust types, with macros doing the wiring. Just like you do with `connect!`.

The ergonomics of the second case seem completely reasonable and familiar.

> It just means, we have to annotate handler functions just to access the docs

This may speak more to my ignorance about how derive macros work, but if you have

```
#[derive(Block)]
#[message_inputs(foo, bar, baz)]
```

why do you have to annotate `foo(...)` at all? Are you not able to confirm that `foo` exists without it in the expansion of `derive(Block)`?

bastibl, 2025-04-25:
If you have this:

```rust
#[derive(Block)]
#[message_inputs(foo)]
struct Foo {
  ...
}

impl Foo {
  pub async fn foo(...) -> Result<Pmt> {
  }
}
```

It expands to something like:

```rust
struct Foo {
  ...
}

impl Foo {
  pub async fn foo(...) -> Result<Pmt> {
  }
}

impl BlockInterface for Foo {
  async fn call(&mut self, id: PortId, pmt: Pmt) {
    match id {
       "foo" => self.foo(pmt).await,
       _ => {},
   }
 }
}
```

This way, the handler function never has to be boxed and the overhead to call a handler should be minimal. The BlockInterface has some more stuff. Everything that the runtime needs to work with it, i.e., it's required to add it as a block to the flowgraph.

I also tend to the second option. I don't like putting longer strings in the attribute macro. Doc comments also have much better IDE support.

metasim, 2025-04-25:
Per my macro ignorance;

If you have this:

```rust
#[derive(Block)]
#[message_inputs(foo)]
struct Foo {
  ...
}

impl Foo {
  /// Access the `foo` property.
  pub async fn foo(...) -> Result<Pmt> {
  }
}
```

Is a derive macro able to capture the doc comment for `foo` so this is possible?

```rust

impl BlockInterface for Foo {
  fn doc_for(id: PortId) -> &'static str {
    match id {
      "foo" => "Access the `foo` property."
    }
  }

  async fn call(&mut self, id: PortId, pmt: Pmt) {
    match id {
       "foo" => self.foo(pmt).await,
       _ => {},
   }
 }
}
```

bastibl, 2025-04-25:
Unfortunately not, that's why you need the extra macro just for getting the docs.