FutureSDR/FutureSDR #198: Remove use of `tracing`'s `release_max_level_debug` and `max_level_debug` features from library crates.

metasim, 2025-04-16:
[According to the `tracing` team](https://discord.com/channels/500028886025895936/627649734592561152/962472729838239805), features such as `max_level_debug`, `release_max_level_info`, etc. should not be used in libraries, as the most restrictive value propagates across the whole application. 

FutureSDR currently declares this:

https://github.com/FutureSDR/FutureSDR/blob/8b8b7739fe2928a551c839258c96b94d98ee1083/Cargo.toml#L99

So if in my application I set `release_max_level_debug`, my debug messages will still get eliminated at compile time because FutureSDR's use of `release_max_level_info` [takes precedence](https://github.com/tokio-rs/tracing/blob/dfc2c8b81889f1bc65f053e574de32ec79b72ce1/tracing/src/level_filters.rs#L69-L83) (a bad design decision on `tracing`'s part IMO... there's [discussion of changing this](https://github.com/tokio-rs/tracing/issues/2081) in the 0.2 release).

I propose removing these flags in the FutureSDR library code.

Related:

* https://discord.com/channels/500028886025895936/627649734592561152/909952192478343180
* https://discord.com/channels/500028886025895936/627649734592561152/1346189126029807817
* https://discord.com/channels/500028886025895936/627649734592561152/1346232581649207346
* https://discord.com/channels/500028886025895936/627649734592561152/909954428684095628

bastibl, 2025-04-25:
I know the discussion and I think they are right that this shouldn't be done (at least in general). But I'd argue that, in this case, you probably do not want any overhead from debug messages in a release build. And I shouldn't be worried that adding debug message might screws release performance.
So the alternative would be to add tracing as a dependency for every project and disable this individually. That's not so nice. And, in addition, most users won't know that they should do this in their own crates.

I'd propose to add two new features (e.g., disable_tracing, disable_debug_in_release) that set these features and enable them by default. This way, advanced users can disable these features. What do you think?

bastibl, 2025-04-25:
I understand. I just think the default should be to have them disabled, like in #199. The default features set the tracing features to disable debug statically but `--no-default-features` removes these features. Is this fine for you?

metasim, 2025-04-25:
> Is this fine for you?

Thanks for asking. :) I have no problem with specifying `default-feature = false` in my dependencies. Much better than dealing with it via a fork.

commit 4dfb50a3 (2025-04-25): tracing: allow configuring features

fixes #198
closed 2025-04-25 by bastibl