FutureSDR/FutureSDR #208: rtlsdr support in android-hw

tkj666, 2025-07-17:
Does android-hw example supports rtlsdr? It seems neither rtl-sdr-rs nor soapy supports opening with fd in args.

bastibl, 2025-07-20:
The example uses the Android SDR libraries from this repository (see build and env scripts):

https://github.com/bastibl/gnuradio-android

Meanwhile, libusb also merged Android support (https://github.com/libusb/libusb/wiki/Android). So the examples could be updated to use the upstream approach (to avoid using patched SDR drivers).

tkj666, 2025-07-20:
Android support in libusb basically involves using `LIBUSB_OPTION_NO_DEVICE_DISCOVERY` and `libusb_wrap_sys_device` to avoid enumerating devices and open device with fd directly. However, the original rtl-sdr driver does not contain function to open a device by fd with `libusb_wrap_sys_device`, so patching the driver is still needed.

bastibl, 2025-07-20:
You're right. But now Android support can be added to the upstream RTL-SDR driver. gnuradio-android used custom hacks that had little chance to get into the upstream driver. I hope that more and more libusb-based drivers will add Android support so that no patch will be needed in the future.